PX4/PX4-Autopilot #17143: FW issue after a yaw impulse

The-Legendary-Kraken, 2021-03-16:
When we fly in FW stabilize and FW acro the UAV turns on the right without any input coming from the RC. The sticks of the RC controller are in the middle.

In order too reproduce the behavior:
1. UAV is in stabilize mode
2. We transition in FW mode increasing the speed
3. turn in left direction with yaw stick the UAV in order to perform a circle at a constant altitude  
4. Put the yaw stick in the middle of the RC controller
5. The UAV instead of going straight start to go right

**Log Files**
https://logs.px4.io/plot_app?log=e5239f62-a575-4f70-b90b-84e6090638f1

In the 3D view at time 4:29:000 is it possible to see that the drone turns right with RC sticks in the middle

The UAV is a VTOL.

The-Legendary-Kraken, 2021-03-17:
I forgot to mention that when I yaw right the drone turns but after I stop it turns left a little bit, so is not a problem of tuning.

fettywapssecondeye, 2021-03-17:
What's the airframe setup like? Looks like it isnt exactly commanded but could you try this with a mode that stabilizes heading as well, like position mode? Motors go crazy after you switch to airplane mode for sure

The-Legendary-Kraken, 2021-03-17:
Yes it is 4 motors tailsitter. Yes after the transition the motors become crazy for few seconds. I tried the FW position mode and the UAV not flyed perfectly straight you can see pos mode here https://logs.px4.io/plot_app?log=76163f0c-f451-4d54-9094-b0b6e229c473 (from 4:12 to 4:19)

fettywapssecondeye, 2021-03-17:
And your airframe is built properly right? Just want to make sure, on hover flight everything seems to be on top of each other but the thrust line of a motor or a couple being off center could cause this behavior as well (or chipped props, ESCs not calibrated properly, control surfaces being out of trim, etc). 

Could be wrong but looked like position mode was converging on stable flight.

closed 2025-12-07 by github-actions[bot]