PX4/PX4-Autopilot #17164: How to deduce the maximum force generated in X, Y and Z directions for the fully-actuated UAV?

daydayupcg137, 2021-03-19:
I read an article recently:《Estimation and Optimization of Fully-actuated Multirotor Platform with Nonparallel Actuation Mechanism》，This article introduces a kind of fully powered UAV, but did not understand how to derive the maximum force in X, Y and Z direction in the paper。The relevant information is as follows:

[image]
[image]
[image]
[image]
[https://www.researchgate.net/publication/321821135_Estimation_and_optimization_of_fully-actuated_multirotor_platform_with_nonparallel_actuation_mechanism](url)

Can a friend provide a solution?

bresch, 2021-03-23:
Hi @daydayupcg137 ,

You first need to compute the `M` matrix by estimating/measuring the rpm-to-force/torque gain of each rotor. Then you isolate `fmax` in the equation and find the maximum of this function given the rpm constraints. You can find the maximum by hand using Lagrange multipliers if you want but also use Python/Matlab to do it.

I'm closing this post because it isn't an issue of the PX4 flight controller software. Please ask those questions on a math/physics forum.

closed 2021-03-23 by bresch

daydayupcg137, 2021-03-24:
Thank you for your reply.But How to  isolate **fmax** in the equation?There are six variables, but there are only five equations.Therefore, I cannot infer the relationship between Fmax and rpm。
the M matrix is shown below:
[image]

In the above case, what is the relation of Fmax(fxmax/fymax/fzmax)  with respect to the tilt Angle alpha and Beta?

I would be very grateful if you could help meThank you very much!