PX4/PX4-Autopilot #17170: Unreachable code block

dgeorge83616, 2021-03-19:
**Describe the bug**
The following block is unreachable because `_v_control_mode.flag_control_attitude_enable` must be both `true` and `false`. It must be `true` to enter the enclosing block and the `false` to enter the `generate_attitude_setpoint()` block.
https://github.com/PX4/PX4-Autopilot/blob/a13459858daec83eefea11e3d9eed94d455de534/src/modules/mc_att_control/mc_att_control_main.cpp#L297-L312

https://github.com/PX4/PX4-Autopilot/commit/84fe64b1c20a4ae8067f04b1e21a961263bac4db

At best, if unreachable the code will not pass serious flight worthiness reviews.

dgeorge83616, 2021-03-19:
Never mind.  I stared and started at that and still missed the `l` vs `t`.

closed 2021-03-19 by dgeorge83616

LorenzMeier, 2021-03-19:
Thanks! It might be worth renaming one of them for that reason.

dgeorge83616, 2021-03-19:
Agree that that would be helpful. I like that PX4 code avoids acronyms, but this case has quite a lot of letters for the amount of information.

dgeorge83616, 2021-03-20:
@LorenzMeier see https://github.com/PX4/PX4-Autopilot/pull/17177, just a small follow-up.