PX4/PX4-Autopilot #17176: Failed to import numpy

Yecgaa1, 2021-03-20:
Failed to import numpy in clion+0.9ToolChain

When I build px4 in Clion with cmake and cywin64
I see the problem

```
[  0%] Generating component_version.json
[  0%] Generating serial_params.c
Scanning dependencies of target drivers_board
[  1%] git submodule src/lib/ecl
[  1%] Generating uORB topic headers
[  1%] Generating mixer_multirotor_6dof.generated.h
[  1%] Generating mixer_multirotor.generated.h
[  1%] Generating mixer_multirotor_normalized.generated.h
[  2%] Generating ../../logs
[  2%] Generating git version header
[  2%] git submodule src/drivers/gps/devices
[  2%] git submodule mavlink/include/mavlink/v2.0
[  2%] Building CXX object boards/px4/sitl/src/CMakeFiles/drivers_board.dir/board_shutdown.cpp.o
Failed to import numpy: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the multiarray numpy extension module failed.  Most
likely you are trying to import a failed build of numpy.
Here is how to proceed:
- If you're working with a numpy git repository, try `git clean -xdf`
  (removes all files not under version control) and rebuild numpy.
- If you are simply trying to use the numpy version that you have installed:
  your installation is broken - please reinstall numpy.
- If you have already reinstalled and that did not fix the problem, then:
  1. Check that you are using the Python you expect (you're using /usr/bin/python3),
     and that you have no directories in your PATH or PYTHONPATH that can
     interfere with the Python and numpy versions you're trying to use.
  2. If (1) looks fine, you can open a new issue at
     https://github.com/numpy/numpy/issues.  Please include details on:
     - how you installed Python
     - how you installed numpy
     - your operating system
     - whether or not you have multiple versions of Python installed
     - if you built from source, your compiler versions and ideally a build log

     Note: this error has many possible causes, so please don't comment on
     an existing issue about this - open a new one instead.

Original error was: No such file or directory

You may need to install it using:
    pip3 install --user numpy

Failed to import numpy: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the multiarray numpy extension module failed.  Most
likely you are trying to import a failed build of numpy.
Here is how to proceed:
- If you're working with a numpy git repository, try `git clean -xdf`
  (removes all files not under version control) and rebuild numpy.
- If you are simply trying to use the numpy version that you have installed:
  your installation is broken - please reinstall numpy.
- If you have already reinstalled and that did not fix the problem, then:
  1. Check that you are using the Python you expect (you're using /usr/bin/python3),
     and that you have no directories in your PATH or PYTHONPATH that can
     interfere with the Python and numpy versions you're trying to use.
  2. If (1) looks fine, you can open a new issue at
     https://github.com/numpy/numpy/issues.  Please include details on:
     - how you installed Python
     - how you installed numpy
     - your operating system
[... 435 lines cut ...]

xtx@DESKTOP-65G5ROL ~
$ which pip
which: no pip in (/usr/local/bin:/usr/bin:/usr/bin:/cygdrive/d/PX4/toolchain/gcc-arm/bin:/cygdrive/d/PX4/toolchain/jdk/bin:/cygdrive/d/PX4/toolchain/apache-ant/bin:/cygdrive/d/PX4/toolchain/genromfs:/cygdrive/d/PX4/toolchain/misc:/usr/lib/lapack)

xtx@DESKTOP-65G5ROL ~
$ which pip3
/usr/bin/pip3

xtx@DESKTOP-65G5ROL ~
$ python --version
Python 3.6.9

xtx@DESKTOP-65G5ROL ~
$ python3 --version
Python 3.6.9

xtx@DESKTOP-65G5ROL ~
$ pip --version
bash: pip: 未找到命令

xtx@DESKTOP-65G5ROL ~
$ pip3 --version
pip 19.2.3 from /usr/lib/python3.6/site-packages/pip (python 3.6)
```

adbidwai, 2021-03-21:
Install numpy for python2 and try again
```
sudo apt-get install python
sudo apt-get install python-numpy
```

Yecgaa1, 2021-03-22:
> Install numpy for python2 and try again
> 
> ```
> sudo apt-get install python
> sudo apt-get install python-numpy
> ```

No!I run the toolChain in Win10

closed 2023-10-12 by Yecgaa1