PX4/PX4-Autopilot #17181: Baro + GPS loss leads to position loss and unstable movements

obicons, 2021-03-21:
**Describe the bug**
When flying with Iris in SITL, I injected a GPS and Barometer failure. This caused the quadcopter to maneuver dangerously.

**To Reproduce**
1. Allow drone to takeoff (any mode, I think)
2. While the drone is ascending, insert a GPS and Barometer failure
3. The drone quickly loses position and altitude

**Expected behavior**
The drone to survive this fault, or trigger a failsafe. 

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=f9a11739-5b30-408b-967e-b0705a937d97

**Drone (please complete the following information):**
SITL with Gazebo Iris.

**Additional context**
N/A

bresch, 2021-03-23:
@julianoes Isn't that covered in CI?

julianoes, 2025-11-09:
Would still be worth testing and fixing this.