PX4/PX4-Autopilot #17187: Make commands not being recognized for fmu rtps builds

blakermchale, 2021-03-22:
**Describe the bug**
In the most recent version of master `make px4_fmu-v3_rtps` is no longer recognized as a command. Only `px4_fmu-v5_rtps` is recognized.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `make px4_fm-v3_rtps` or `make check_rtps`

**Expected behavior**
This should build the binaries with RTPS for fmu-v3.

**Log Files and Screenshots**
[image]

**Drone (please complete the following information):**
Not applicable.

**Additional context**
Not applicable.

LorenzMeier, 2021-03-22:
We likely disabled this as FMUv3 is a very old target (2015/16-ish), running out of flash space and RAM and most of us use a Pixhawk 4 or newer. Is it the only hardware option you have?

blakermchale, 2021-03-22:
I'm part of an undergraduate student organization where we have designed our own drones around the 3DR Pixhawk Mini. We purchased them a while ago so it is currently our only option for our drones. It seems like a recent update disabled this since we had been using a version of master from January 22, 2021 and were able to build this version fine.

blakermchale, 2021-07-15:
This can be implemented by adding your own cmake files with RTPS enabled.

closed 2021-07-15 by blakermchale