PX4/PX4-Autopilot #17188: CUAV nano compile and upload v1.12.0-beta1 program error

mengchaoheng, 2021-03-22:
**Describe the bug**
The initial bell（STARTUP_TUNE） is incomplete, the output of the servo is wrong after startup, and the safety switch does not work.

**To Reproduce**
Steps to reproduce the behavior:
1. make px4_fmu-v5_default
2. make px4_fmu-v5_default upload
3. See error

**Expected behavior**
The tune_control play -t $STARTUP_TUNE command will work. After starting, press the safety switch and the servo output will appear.

**Log Files and Screenshots**
[image]
[image]

**Additional context**
But when I download the beta firmware in QGC, it can run normally. I don’t know why the firmware compiled locally has problems.

mengchaoheng, 2021-03-22:
https://user-images.githubusercontent.com/43166007/111938902-f23b4500-8b05-11eb-9367-f6a2cfaec961.mp4

----------------------------------------------------------
This is the wrong initial ringtone

mengchaoheng, 2021-03-23:
@LorenzMeier

mengchaoheng, 2021-03-23:
https://github.com/PX4/PX4-Autopilot/issues/16883, see also

mengchaoheng, 2021-03-24:
Additional context
But when I download the beta firmware in QGC, The safety switch is also wrong. It is open sourced but cannot be closed.

LorenzMeier, 2021-03-24:
Safety now defaults to off - just added that to the release notes: https://github.com/PX4/PX4-Autopilot/releases/tag/v1.12.0-beta1

mrpollo, 2021-04-28:
@mengchaoheng did @LorenzMeier comment help you? are you still experiencing this issue?

mengchaoheng, 2021-04-28:
yes, it work now.

| |
蒙超恒
|
|
***@***.***
|

签名由 网易邮箱大师 定制

On 04/29/2021 00:13, Ramon Roche wrote:

@mengchaoheng did @LorenzMeier comment help you? are you still experiencing this issue?

mrpollo, 2021-04-28:
Thanks @mengchaoheng 👍

closed 2021-04-28 by mrpollo