PX4/PX4-Autopilot #17192: compass loss causes unsafe maneuvers

obicons, 2021-03-22:
**Describe the bug**
During the landing sequence, a compass failure leads to a position loss and the vehicle jerks violently. Tested on recent commit. 

**To Reproduce**
1. Takeoff
2. Land
3. While landing, fail every compass instance

**Expected behavior**
The vehicle to proceed through the landing sequence.

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=b9da48e7-9b6a-4a87-a9b0-4d1ffb8d4e26

**Drone (please complete the following information):**
SITL Gazebo Iris

**Additional context**
N/A

bresch, 2021-03-22:
There are a few strange things that would require further investigations:
- The mag data already stops before takeoff (are you sure you only stopped it during landing?)
- The GNSS vel/pos fusion only seem to start after the mag data stopped
- When the GNSS fusion resets, it has a huge value (was the origin incorrect?)
- The position setpoint isn't exactly adjusted during the reset. It could be a race condition because it looks like the EKF selector switches at the same time.
[image]

priseborough, 2023-02-07:
The issue in this log is not compass loss, it is an invalid GPS message that results in the EKF resetting it's position to a 0 lat,lon.

closed 2023-02-07 by priseborough