PX4/PX4-Autopilot #17199: RPi3: LSM9DS1 ERROR: Accel #0 fail: TIMEOUT!, vehicle_magnetometer lost

Pskoog3, 2021-03-22:
With the latest build the accel/gyro/mag sensor of LSM9DS1 gets calibrated with the correct sensitivity in QGroundControl, but now getting a vehicle_magnetometer lost error as indicated in the following log. Moreover, in QGroundControl it constantly complains for Accel 0 fail: TIMEOUT! By the way I am using a custom sensor board with Raspberry Pi 3.

$ sudo ./bin/px4 -s px4_rpi.config
INFO [px4] mlockall() enabled. PX4's virtual address space is locked into RAM.
INFO [px4] assuming working directory is rootfs, no symlinks needed.

| ___ \ \ \ / / / |
| |/ / \ V / / /| |
| __/ / \ / /| |
| | / /^\ \ ___ |
_| / / |_/

px4 starting.

INFO [px4] Calling startup script: /bin/sh px4_rpi.config 0
px4_rpi.config: 8: px4_rpi.config: uorb: not found
ERROR [param] Parameter MAV_BROADCAST not found.
INFO [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
lsm9ds1 #0 on SPI bus 0 (devid=0x0)
lsm9ds1_mag #0 on SPI bus 0 (devid=0x1)
ms5611 #0 on I2C bus 1
INFO [pca9685_pwm_out] running on I2C bus 1 address 0x40
INFO [commander] LED: open /dev/led0 failed (22)
INFO [mavlink] mode: Normal, data rate: 1000000 B/s on udp port 14556 remote port 14550
INFO [mavlink] mode: Normal, data rate: 2880 B/s on /dev/ttyUSB-Telemetry @ 57600B
INFO [logger] logger started (mode=all)
WARN [logger] battery_status not published. Logging anyway
INFO [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO [px4] Startup script returned successfully
pxh> ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 0 -> 21
ERROR [sensors] Accel #0 fail: TIMEOUT!
ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 23 -> 25
ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 27 -> 29
ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 31 -> 33
ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 35 -> 37
ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 39 -> 41
ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 43 -> 45
ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 47 -> 49
ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 51 -> 53
...

Pskoog3, 2021-03-23:
The latest log might show the problem...
https://review.px4.io/plot_app?log=58273f87-acf8-4020-8e7b-b5f0ae6327a1

dagar, 2021-03-26:
Hi @Pskoog3,

Can you run `uorb top`? I'm looking for the publication rate of sensor_accel, sensor_gyro, etc. It sounds like the `LSM9DS1` is not working right at all.

Can you describe everything that's on this board? For testing the closest thing I have is a Navio2.

dagar, 2021-03-26:
@Pskoog3 please try master now that https://github.com/PX4/PX4-Autopilot/pull/17239 has merged.

Pskoog3, 2021-03-29:
Still same issue... sensors seems to work fine and gets calibrated but still having the following error messages in console and Accel 0 fail: TIMEOUT! constantly coming up!

ERROR [sensors] Accel #0 fail: TIMEOUT!
ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 23 -> 25

dagar, 2021-03-29:
Thanks for the update. Can you grab a fresh log with current master to share? https://logs.px4.io/
Are you running px4 as root? Maybe capture a screenshot of `top` (or `htop`) while px4 is running to get a better idea what's happening both on the system and within px4.

I use htop to see the individual threads within px4.

Pskoog3, 2021-03-30:
@dagar here is the fresh log with recent master... 
https://logs.px4.io/plot_app?log=e423f38f-4090-4a16-bf73-d3aeab883f5e

yes, running as root... 
$ sudo ./bin/px4 -s px4_rpi.config

screenshot of <top> with px4 running...
[image]

screenshot of <htop>...
[image]

screenshot of console...
[image]

screenshot of QGroundControl messages...
[image]

closed 2025-11-06 by hamishwillee