PX4/PX4-Autopilot #17212: Flashing nxp_fmuk66-v_rtps or nxp_fmuk66-v_default causes continuous self-rebooting

KatrinMoritz, 2021-03-23:
**Describe the bug**
Flashing the nxp_fmuk66-v3_rtps.px4 or nxp_fmuk66-v3_default.px4 via QGC causes that the FMU is always rebooting and not starting correctly. It reboots before the communication with QGC is started. Therefore the FMU is with this version not usable. This behavior can be observed from the commit e7722b11eb. In the commit before (c1c0a62be5) the firmware works as expected.

**To Reproduce**
Steps to reproduce the behavior:
1. Build the firmware on the status of commit e7722b11eb
2. Uploaded firmware (.px4) via QGC
3. See error: directly rebooting after every reboot

**Expected behavior**
Blue light is blinking slow. FMU connects to QGC after flashing.

**Log Files and Screenshots**
I want to upload a video but the file is too big.

**Drone (please complete the following information):**
- Rover Setup with NXP Cup Rover
- RDDRONE-FMUK66

KatrinMoritz, 2021-03-23:
@96moustafa @dk7xe

dk7xe, 2021-03-23:
@dagar @davids5 fyi

davids5, 2021-03-23:
@dk7xe can you get a console on the debug connector? Is it a hardfault in the PWM module?

dagar, 2021-03-23:
This should be fixed in master as of https://github.com/PX4/PX4-Autopilot/commit/9b1cf98474b271ef402dd3d7dd47acd581acc1d7.

KatrinMoritz, 2021-03-24:
I tested the new master with both versions (default and rtps). The rebooting issue is fixed with 9b1cf98, @dagar.

closed 2021-03-24 by KatrinMoritz