PX4/PX4-Autopilot #17214: Help with reference frames

int-smart, 2021-03-23:
Hi
I am using a thoneflow with an external lidar module TFMini Plus for Position hold. 
This is with reference to this: https://github.com/PX4/PX4-Autopilot/blob/46f0388fc778d63ff1ded2125c76e006d3690f74/src/modules/local_position_estimator/sensors/flow.cpp#L88

1. The link is for the calculation of delta_b which I feel is the linear velocity of the drone. The linear component is being derived from the flow values and the angular velocities from the gyro. I, however am not able to find any information on how this formula is derived. Is there any documentation (research paper or so) on this. I found some related literature in Angular Rate Compensation section here https://people.inf.ethz.ch/pomarc/pubs/HoneggerICRA13.pdf, however the formula requires focal length and other parameters that are not available to us on cameras such as thoneflow.

2. I am also confused on the reference frame. The documentation states:
The output of the flow when moving in different directions must be as follows:

Vehicle movement | Integrated flow
-- | --
Forwards | + Y
Backwards | - Y
Right | - X
Left | + X

which says that the +Y of the flow should be forward which I believe is the reason for
body frame velocity

           delta_b(
    		+(flow_y_rad - gyro_y_rad) * d,
    		-(flow_x_rad - gyro_x_rad) * d,
    		0);
as the +Y of the flow corresponds to +X of the drone and +X of the flow corresponds to -Y of the drone. However, flow_y_rad and gyro_y_rad are coming from two reference frames in case of sensors such as thoneflow which dont have a gyro on board. In such a case gyro +X axis would be aligned with +Y of the flow. Is this understanding correct.

3. The documentation states _The “zero” rotation is when the sensor board and vehicle X-axes are aligned (i.e. the “front” of the vehicle and the board are in the same direction), with rotations increasing in a clockwise direction. PMW3901 modules have a small notch indicating the direction of the back of the board. The diagram shows the orientation of the Bitcraze board to the vehicle that corresponds to SENS_FLOW_ROT=0 (note the notch at the back)._
Does this still say that +Y of the flow should be aligned to forward of the drone.

Thanks

bresch, 2021-03-24:
The definition used is that a positive flow is created by a position rotation around an axis.

This means that a positive rotation around the X axis creates a positive X flow; and this is the same flow that is generated when the drone translates in the left direction.

Same for example for +Y. This is a positive rotation around the Y body axis, which generates the same flow as when moving forward.

int-smart, 2021-03-24:
Thanks for the explanation. I just had one more question: the gyro values (how much has the drone rotated on its axes) are being subtracted from the flow values as flow values include the rotation of the drone on its own axes apart from the rotation of the drone with respect to the environment. Is this understanding correct?

bresch, 2021-03-25:
> rotation of the drone on its own axes apart from the rotation of the drone with respect to the environment

I'm not sure if I get that sentence, but to answer the question: yes, the EKF compensates for the apparent flow due to the angular velocity of the drone. It is done here: https://github.com/PX4/PX4-ECL/blob/6f2dec726adeedcb53aad9a77a8fcdfc8f7b3054/EKF/control.cpp#L375-L376

int-smart, 2021-03-25:
Thanks for the response. I apologise if I made the question complicated. I am interested in knowing what delta_b here: https://github.com/PX4/PX4-Autopilot/blob/46f0388fc778d63ff1ded2125c76e006d3690f74/src/modules/local_position_estimator/sensors/flow.cpp#L88 represents. How do we arrive at this formula. I was assuming delta_b to be the rotation of the drone with respect to the environment which is why gyro values (how much has the drone rotated on its axes) are being subtracted from the flow values as flow values include the rotation of the drone on its own axes and the rotation of the drone with respect to the environment.

bresch, 2021-03-25:
@int-smart I'm not familiar with the Local position estimator module, sorry. This is also not maintained anymore so please use EKF2 if possible (and check the code by following the links I posted above).