PX4/PX4-Autopilot #17216: Some old params in v1.12.beat1 and A short time pause at before transfer(MC to WING) in VTOL plane

Kim-zhi-jiang, 2021-03-24:
Hello there

I can't find some old params in v1.12.beat1 and A short time pause at before transfer(MC to WING) in VTOL plane.

https://github.com/PX4/PX4-Autopilot/tree/v1.12.0-beta1/src/modules/sensors/

PWM_MIN ---I don't know
PX4-Autopilot/src/modules/sensors/pwm_params_main.c Not file

PWM_MAX ---I don't know
PX4-Autopilot/src/modules/sensors/pwm_params_main.c Not file

PWM_DISARMED ---I don't know
PX4-Autopilot/src/modules/sensors/pwm_params_main.c Not file

PWM_AUX_DISARMED ---I don't know
/PX4-Autopilot/src/modules/sensors/pwm_params_aux.c Not file

PWM_RATE ---OK Default 400 
PX4-Autopilot/src/modules/sensors/pwm_params_main.c Not file

FW_RSP_OFF ---OK Deletied
PX4-Autopilot/src/modules/fw_att_control/fw_att_control_params.c

MPC_JERK_MIN ---OK Deletied
/PX4-Autopilot/src/modules/mc_pos_control/mc_pos_control_params.c

[image]

jinger26, 2021-03-24:
@sfuhrer @RomanBapst  can you take a look at the transition issue here?

jinger26, 2021-03-24:
@dagar reminder to respond with regards to parameters translation per dev-call, thank you!

closed 2021-04-05 by Kim-zhi-jiang