PX4/PX4-Autopilot #17220: Motors spin up to ~1500 pwm when vehicle is armed. [Control Allocation V1.12.0Beta]

akarshgopal, 2021-03-24:
**The bug**
Motors spin up to ~1500 pwm when vehicle [ Hexacopter Control Allocation V1.12.0Beta] is armed.

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on
2. Arm via RC or GQC
3. See motors spin to ~1500 pwm

**Expected behavior**
Motors spin at idle speed (1075 pwm default)

**Log Files and Screenshots**
https://review.px4.io/plot_app?log=f2aa0437-ef1d-416c-aa16-5c64591922fa

**Drone (please complete the following information):**
- Pixhawk 4, hexacopter
- px4 v1.12.0 beta1
- Control allocation hexacopter

**Additional context**
- Tried the same setup with normal hexacopter airframe, the issue did not show up.
- CA_ACTX parameters have MIN 0, MAX 1
- Changing PWM_MAIN_MIN did not affect the issue
- Throttling up and down shows no changes in motor PWM

akarshgopal, 2021-04-10:
Did some digging and found out the following:
- actuator_controls_4 and actuator_controls_5 uOrb messages seem to function normally, i.e the output messages of the control allocation module are in order.

- actuator_outputs uOrb topic is problematic: 
   - first 8 outputs stuck at ~1500 pwm on arming, with no changes due to throttle, or stick inputs.
   - actuator_outputs message is not problematic in SITL  (could be due to the routing through pwm_out_sim instead of pwm_io?)
   - so seems like pwm_io is the source of the bug?

akarshgopal, 2021-04-10:
Some more observations:
- pwm_out driver was not running by default. Not sure if that is intentional?
- actuator_controls_4 and actuator_controls_5 (which afaik is used for CA) don't seem to be subscribed to or used in px4io.cpp

jlecoeur, 2021-04-11:
@akarshgopal control allocation is supported on FMU only, it is not supported on IO. IO support will come when we have removed mixing from the IO processor.

Try connecting your motors to FMU outputs (and load the appropriate mixer on FMU.

akarshgopal, 2021-04-11:
> @akarshgopal control allocation is supported on FMU only, it is not supported on IO. IO support will come when we have removed mixing from the IO processor.
>
> Try connecting your motors to FMU outputs (and load the appropriate mixer on FMU.

Okay thanks for letting me know. Anyway I can help speed up IO support? 
My objective with CA is to fly a tilting-hexacopter (overactuated) so really need 6 motors + 6 servos. Any way I can work around the lack of IO support and get 12 pwm channels out?

Does the 6003_hexa_x_ctrlalloc airframe config file not load the appropriate mixer for the FMU? If not what commands do I need to run or what modifications do I need to make?

jlecoeur, 2021-04-12:
The relevant PR is here https://github.com/PX4/PX4-Autopilot/pull/16444

Alternatively, you can have a quick solution/hack by modifying a bit the control allocation code. Actuator outputs are now scaled and sent on topics actuator_controls_4 and actuator_controls_5. Rename actuator_controls_5 to actuator_controls_0 and you can access outputs 8 to 15 in PX4IO.

akarshgopal, 2021-04-12:
Awesome, thanks!

Great to hear there might be a workaround for the time being, will try it out! 

Should we close this issue btw?

akarshgopal, 2021-06-22:
For reference, 
I was able to work around this using actuator_controls_2 to access px4io output pins. Needed to use a summing mixer. 
Working branch with the above changes: https://github.com/akarshgopal/Firmware/tree/saxion-m4e-tiltrotor

closed 2021-06-22 by akarshgopal

ErcBunny, 2022-03-03:
Hi, I am using stable release v1.12.3. It seems that this issue is still not fixed. I can get the right `actuator_controls_4` but `actuator_outputs` still step up to 1512 when armed, and its data do not respond to changes of `actuator_controls_4`. Airframe: `6003_hexa_x_ctrlalloc`.

BTW, I notice the line `mc_rate_control stop` in `rc.ctrlalloc`, but `mc_rate_control status` returns `running`. This behavior is observed in SITL. I am not sure if this occurs on real hardware too.