PX4/PX4-Autopilot #17229: Guide for copying `launch` folder to `~/catkin_ws/px4/src`

bigbellmercy, 2021-03-25:
Hi~ I have a suggestion for document improvement for ROS related.

I recently get to know that the `launch` folder should be copied to `~/catkin_ws/src/px4` after it is created first, in order to use the launch files for ROS.

Is it automatically created and copied by some script? But in my case it didn't.

When I searched internet there is no clue for such thing.

Thanks.

PX4 has dedicated repositories for developer documentation (https://github.com/PX4/Devguide) and user documentation (https://github.com/PX4/px4_user_guide).

Thanks!

Jaeyoung-Lim, 2021-03-26:
@bigbellmercy You need to follow the documentation in http://docs.px4.io/master/en/simulation/ros_interface.html, where the px4 autopilot repository should not be part of the catkin workspace but outside of it.
e.g.
```
cd <PX4-Autopilot_clone>
DONT_RUN=1 make px4_sitl_default gazebo
source ~/catkin_ws/devel/setup.bash    # (optional)
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
roslaunch px4 posix_sitl.launch
```
This is because PX4 is not dependent on catkin or ROS and it does not follow some of the assumptions that catkin is making regarding install paths

bigbellmercy, 2021-03-26:
I see. That method works well!  Thank you.

closed 2021-03-26 by Jaeyoung-Lim