PX4/PX4-Autopilot #17232: CRSF telemetry should also work on modalai/fc_v1

benjinne, 2021-03-25:
**Describe the bug**
https://github.com/PX4/PX4-Autopilot/blob/5a1eb3f9eff6510c0eb6f93b9338c65832b80150/src/drivers/rc_input/RCInput.cpp#L627
This line only allows CRSF telemetry to work with the omnibus F4SD board, however I've removed it and verified it also works with the modalai/fc_v1 board as well. What is the macro needed for the modalai/fc_v1 board?

bkueng, 2021-03-26:
`CONFIG_ARCH_BOARD_MODALAI_FC_V1`. A bit better would be to add a `BOARD_SUPPORTS_RC_SERIAL_PORT_OUTPUT` define to the board config and check for that in RCInput.cpp.

benjinne, 2021-03-29:
@modaltb could you take a look at this for me?

modaltb, 2021-03-29:
@benjinne what are you using here?  This isn't something we're testing against yet, but if it works, and we can do it in a less board specific way as Beat mentioned above, it might be better!

benjinne, 2021-03-29:
@modaltb I'm using crossfire telemetry with my RC transmitter. It's very similar to frsky telemetry except crossfire RC control and telemetry run on one uart since crossfire receivers use both RX and TX and run on the CRSF protocol. I've verified this works by connecting a crossfire receiver to J1004 (RC Input port) with TX going to Sbus and RX going to TX. All I had to do was remove the macro to only allow CRSF telemetry for the Omnibus FC, and upload my custom build to the FC. I agree with Beat this should be done in a less board specific way. Let me know if you have a solution

closed 2021-04-07 by benjinne