PX4/PX4-Autopilot #17235: OrangeCube: Check for stack overflow on each function call causes hardfault

dgeorge83616, 2021-03-25:
**Describe the bug**
Debug serial console at boot:
```
[2021-03-25 13:50:01] [boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[2021-03-25 13:50:01] [boot] Fault Log is Armed
[2021-03-25 13:50:01] arm_hardfault: Hard Fault:
[2021-03-25 13:50:01] arm_hardfault:   IRQ: 3 regs: 0x37ffff04
[2021-03-25 13:50:01] arm_hardfault:   BASEPRI: 000000f0 PRIMASK: 00000000 IPSR: 00000003 CONTROL: 00000000
[2021-03-25 13:50:01] arm_hardfault:   CFAULTS: 00000400 HFAULTS: 40000000 DFAULTS: 00000000 BFAULTADDR: 00000000 AFAULTS: 00000000
[2021-03-25 13:50:01] arm_hardfault: PANIC!!! Hard fault: 40000000
[2021-03-25 13:50:01] up_assert: Assertion failed at file:armv7-m/arm_hardfault.c line: 134 task: l"
```

**To Reproduce**
See below.

**Expected behavior**
Normal boot.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
See [slack thread here](https://px4.slack.com/archives/C0W2KUFFT/p1616547404075700). @dagar suggested enabling stack checking to help diagnose, but first, tried updated master as discussed [here](https://px4.slack.com/archives/C50RWGQMT/p1616628176118100). 

Decided to verify stack checking would work on master, without any changes, other than enabling the checking and discovered the above.

**Drone (please complete the following information):**
- N/A
Cube Orange sitting on a Kore carrier on desktop. RC to pilot console, USB serial for both QGC and SEGGER j-link for NuttX debug console.

**Additional context**
What follows are captures from tmux log leading to discovering the fault.

After attempting to merge #16444 into master branch, and failing to cleanly resolve conflicts, I decided to just enable stack checking on the master branch.
```
dan@ubuntu:~/projects/px4-community$ git reset --hard HEAD                                                                                                                                                                                                                         
HEAD is now at beadae6dfc process_sensor_caldata.py: Fix 2nd barometer plot                                                                                                                                                                                                        
dan@ubuntu:~/projects/px4-community$ git status                                                                                                                                                                                                                                    
On branch master                                                                                                                                                                                                                                                                   
Your branch is up to date with 'origin/master'.                                                                                                                                                                                                                                    
                                                                                                                                                                                                                                                                                   
nothing to commit, working tree clean                                                                                                                                                                                                                                              
dan@ubuntu:~/projects/px4-community$ git log                                                                                                                                                                                                                                       
commit beadae6dfc032079bd1d0d4300a8854e45eae39f (HEAD -> master, origin/master, origin/HEAD)                                                                                                                                                                                       
Author: Serhat Aksun <serhat.aksun@maxwell-innovations.com>                                                                                                                                                                                                                        
Date:   Mon Mar 22 14:42:35 2021 +0300
```
`distclean` the workspace.
```
dan@ubuntu:~/projects/px4-community$ make distclean                                                                                                                                                                                                                                
Cleared directory 'Tools/flightgear_bridge'                                                                                                                                                                                                                                        
warning: Could not unset core.worktree setting in submodule 'Tools/flightgear_bridge'                                                                                                                                                                                              
Cleared directory 'Tools/jMAVSim'                                                                                                                                                                                                                                                  
warning: Could not unset core.worktree setting in submodule 'Tools/jMAVSim'                                                                                                                                                                                                        
Cleared directory 'Tools/jsbsim_bridge'                                                                                                                                                                                                                                            
warning: Could not unset core.worktree setting in submodule 'Tools/jsbsim_bridge'                                                                                                                                                                                                  
Cleared directory 'Tools/sitl_gazebo'                                                                                                                                                                                                                                              
warning: Could not unset core.worktree setting in submodule 'Tools/sitl_gazebo'                                                                                                                                                                                                    
Cleared directory 'boards/atlflight/cmake_hexagon'
warning: Could not unset core.worktree setting in submodule 'boards/atlflight/cmake_hexagon'
Cleared directory 'mavlink/include/mavlink/v2.0'
Submodule 'mavlink/include/mavlink/v2.0' (https://github.com/mavlink/c_library_v2.git) unregistered for path 'mavlink/include/mavlink/v2.0'
Cleared directory 'platforms/nuttx/NuttX/apps'
Submodule 'platforms/nuttx/NuttX/apps' (https://github.com/PX4/NuttX-apps.git) unregistered for path 'platforms/nuttx/NuttX/apps'
Cleared directory 'platforms/nuttx/NuttX/nuttx'
Submodule 'platforms/nuttx/NuttX/nuttx' (https://github.com/PX4/NuttX.git) unregistered for path 'platforms/nuttx/NuttX/nuttx'
Cleared directory 'platforms/qurt/dspal'
warning: Could not unset core.worktree setting in submodule 'platforms/qurt/dspal'
Cleared directory 'src/drivers/gps/devices'
Submodule 'src/drivers/gps/devices' (https://github.com/PX4/PX4-GPSDrivers.git) unregistered for path 'src/drivers/gps/devices'
Cleared directory 'src/drivers/uavcan/libuavcan'
[... 149 lines cut ...]
Date:   Thu Mar 25 00:25:29 2021 -0400

    Update submodule GPSDrivers to latest Thu Mar 25 00:41:54 UTC 2021

commit 98e1e25d93217f76b567c9fe2401e62ed132fd28
Author: PX4 BuildBot <bot@px4.io>
Date:   Thu Mar 25 00:41:48 2021 +0000

    Update submodule sitl_gazebo to latest Thu Mar 25 00:41:48 UTC 2021
    
[... cut ...]
        - Changes: https://github.com/PX4/PX4-SITL_gazebo/compare/1f339cdf5c5da27097e58ec2beb2793bf6057c7d...3e5fed04d8e574b10e17e446d2938346bc6152ca
    
        3e5fed0 2021-03-22 JaeyoungLim - Multiple airspeed sensors in SITL with multi HIL_SENSOR instances (#731)

commit 7f2566dd8a2158325a6ad57dcb5250e1056165e9
Author: David Sidrane <David.Sidrane@Nscdg.com>
Date:   Wed Mar 24 16:29:40 2021 -0700

    NuttX STM32H7 16 bit SPI fixes + ADIS16470 updates for proper 16 bit mode in burst

commit 92910de81d66480a9f22cd9237200fbacac10122

```

julianoes, 2023-04-03:
I'm just trying to reproduce this but I'm not sure how this would possibly fit into flash? :astonished: 

```
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: cubepilot_cubeorange_default.elf section `.text' will not fit in region `FLASH'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 938840 bytes
```

julianoes, 2025-11-05:
No more follow up.

closed 2025-11-05 by julianoes