PX4/PX4-Autopilot #17237: Cuavx7 top 8 PWM channels have no output

CUAVcaijie, 2021-03-26:
**Describe the bug**
CUAVx7 MAIN channel has no output 

**To Reproduce**
1.Firmware version: a12442654138c88c796f9a655907f78fc258d04d
2. Set the model to 4001 
[image]

3. Calibrate all sensors 
4. Setting parameters: 
5. Restart 
6. Connect to the remote control

Test Results:
CAUVx7:
    PWM output before unlocking: 
[image]

    Unlock PWM output:
[image]

Analysis: No output from MAIN interface, normal output from AUX interface 

Use CUAVv5 for comparison: 
    PWM output before unlocking: 
[image]

    Unlock PWM output:
[image]

CUAVcaijie, 2021-03-26:
parameter settings:
CBRK_USB_CHK = 197848
CBRK_SUPPLY_CHK = 894281
RC_MAP_AUX1 = Channel 6
RC_MAP_AUX2 = Channel 7

dagar, 2021-03-26:
Thanks for reporting, I think we have a regression.

closed 2021-06-07 by CUAVcaijie