PX4/PX4-Autopilot #17238: Multicopter yaw as gimbal

holmnikolaj, 2021-03-26:
I have a multicopter with a camera on a one axis gimbal (for pitch). To control my ROI i want to use the multicopter yaw, such that instead of yaw to whatever the mission (etc.) decides, then control the mc yaw based on what the gimbal controller output.

This could be possible by modifying vmount to not account for drone attitude, and use vmount output as yaw reference. This should of cause be configured by a parameter (it could be MNT_MODE_OUT)

dagar, 2021-03-26:
Interesting idea, I'd be happy to discuss on a call if you'd like to try and implement this.

holmnikolaj, 2021-03-29:
I would. As I see it there are some issues that would be good to discuss:

 - What is the best way to get the output from vmount to mc_att_control? We could use mount_orientation.msg but that might be a problem if the gimbal publishes to that topic. Alternatively mc_att_control could subscribe to vehicle command, we could create a new uorb topic for it or make it an addition to mount_orientation. 

 - In vmount the output is either aux or mavlink. It should still be possible to choose how to output the vmount pitch and roll when using the yaw part in mc_att_control. 

 - Are there some cases where we have to abandon this scheme (even when the user want it). Do we have indications of limited control robustness that we risk worsening by yaw'ing "arbitrarily"? 

@dagar when do you have time to discuss? I prefer somewhere bewteen 8-16 CEST.

holmnikolaj, 2021-06-09:
@dagar - I'm about to implement this. Do you still want to discuss on a call beforehand?

holmnikolaj, 2021-06-22:
[https://github.com/PX4/PX4-Autopilot/pull/17789](url)

closed 2025-12-05 by github-actions[bot]