PX4/PX4-Autopilot #17240: Collision prevention does not work with firmware 1.12.0-beta1 and master

saengphet, 2021-03-26:
**Describe the bug**
hello there, I installed the lightware lidar LW20/C and Stereo camera D435i (runningPX4 avoidance for stereo) in front of Quadrotor for collision prevention. It works fine with firmware version v1.11.3. both lidar and stereo camera However, after I  upgraded the firmware to v1.12.0-beta1 ( tried master too on 26/03/2021). Even I did not touch any parameter relate to collision prevention, it does work as expected. The vehicle did not stop and continue going forward when the pilot moves the pitch stick input forward in Position mode.

**To Reproduce**
Steps to reproduce the behavior:
1. install LW20/C set -R 0 (forward) and D435i running with px4 avoidance in companion computer, set CP_DIST = 5 m.
2. takeoff in position mode
3. move pitch stick input forward to the obstacle
4. See an error, the vehicle does not stop at 5 m and continue moving forward to the obstacle

**Expected behavior**
the vehicle stop at 5 m even pilot moves pitch stick input forward in position mode to the obstacle

**Drone (please complete the following information):**
- Quadrotor
- Pixhawk 4 (also tried the cube orange)
- lidar lightware LW20/C (I try ll40ls i2c also, the result is the same)
- Stereo camera D435i with companion computer

**Additional context**
I have tried it in gazebo simulation with h480 airframe. The result is the same. the vehicle does not stop as expected.

dagar, 2021-03-26:
What do you have `MPC_POS_MODE` set to? Collision prevention is currently only available with MPC_POS_MODE=3 (Smooth position control with maximum acceleration and jerk limits), and not the new default MPC_POS_MODE=4 (Smooth position control where sticks map to acceleration and there's a virtual brake drag).

FYI @MaEtUgR

saengphet, 2021-03-27:
> What do you have `MPC_POS_MODE` set to? Collision prevention is currently only available with MPC_POS_MODE=3 (Smooth position control with maximum acceleration and jerk limits), and not the new [...]

I have already tested it by setting MPC_POS_MODE to 3: Smooth position control (Jerk optimized). I switch from "altitude mode" to "position mode", then the FCU seems stuck momentarily and back to normal, meanwhile the GPS LED changes from green to blue and Cannot switch to Position mode again. It is very strange behavior I have not ever seen. It likes the nav/gps or position status stop working after change to position mode. The FCU also disconnects from the companion computer at the same time.

dagar, 2021-03-27:
Can you share the flight log?

saengphet, 2021-03-28:
@dagar 

**- Firmware v1.11.3 - CP is OK**
https://review.px4.io/plot_app?log=f8784279-4476-4294-a658-3d6839603464

**- Firmware master CP is NOT OK**
https://review.px4.io/plot_app?log=1d898f6b-c164-4a81-b687-a3e593011595

https://review.px4.io/plot_app?log=e79f46ae-8c9b-4c28-a2e8-acf6a3c4dd02

saengphet, 2021-04-02:
@dagar  any update? Thanks

saengphet, 2021-05-25:
@dagar  I have not tested these again using firmware beta5. Have this problem been solved?

barrenwufffet, 2022-01-25:
@saengphet  Any resolve on this by chance? Experiencing the exact same issue and having issues troubleshooting

dagar, 2022-01-25:
@barrenwufffet did you check MPC_POS_MODE set to 3?

barrenwufffet, 2022-01-27:
Yes I have tried using MPC_POS_MODE set to 3 and 4 and the behavior is the exact same across both