PX4/PX4-Autopilot #17244: X2.1 Mag not showing up

nbort, 2021-03-26:
**Describe the bug**
The X2.1 (not -777) onboard mag is not showing up in PX4 v1.11.3

**To Reproduce**
Steps to reproduce the behavior:
1. Connected a new X2.1, preinstalled with Ardupilot, to QGC. Verified that the mag was seen 
2. Used QGC to change the firmware to PX4. Automatically loaded px4_fmu-v3 (surprised it did not load v2)
3. "uorb top" did not show the mag or either IMU
4. Performed the x2.1 SD bootloader upgrade
5. Used QGC to change the firmware. Automatically loaded auav-x21_default.px4
6. "uorb top" shows both accels, both gyros but not the mag
7.  Completed the QGC compass cal but no compass was calibrated (all mag IDs and mag_comp still zero)
8.  Used QGC to change the firmware. Manually loaded mro_x21_default.px4 (1.11.3)
9. "uorb top" shows both accels, both gyros but not the mag
10. Completed the QGC compass cal but no compass was calibrated (all mag IDs and mag_comp still zero)
11. Repeated the entire process on a second x2.1 autopilot

**Expected behavior**
I expect to see the onboard mag in the mavlink console and expect to see it show up as calibrated after performing a QGC compass calibration

**Drone (please complete the following information):**
x2.1 Autopilot. Standard Plane vehicle type. The above testing was performed with the autopilot connected to a Windows computer via USB with no other autopilot connections.

**Additional context**
This follows testing with a third x2.1 in which I had the same problem but was able to see and calibrate external mags. On an x2.1-777 I can see the mag (but it fails to do other things, like PPM in)

dagar, 2021-03-26:
Do you happen to know if the only internal mag is supposed to be from the mpu9250?

https://github.com/PX4/PX4-Autopilot/blob/a260ddd8ff4b91488271c5ae13bac9065467309d/boards/mro/x21/src/spi.cpp#L39-L43

In the updated mpu9250 we no longer run the magnetometer by default because it interferes (mildly) with the high rate raw data output. If that really is the only onboard magnetometer I can enable it.

nbort, 2021-03-26:
Yes, I am referring to the mpu9250. It is the only mag on the board. My airplane will have an external mag but I do like having the internal mag, mainly for the preflight consistency checks. It is interesting that with the mro x2.1-777, I do see the mpu9250 mag with 1.11.3.

dagar, 2021-03-26:
Can you give https://github.com/PX4/PX4-Autopilot/pull/17246 a try?

If you take a look at `work_queue status` and `uorb top sensor` you should see the impact of the mag on mpu9250 scheduling.

nbort, 2021-03-27:
Thanks! I verified that it worked, both forked from the master and from 1.11.3. I just did basic functional checks - the mag shows up, the field reading make sense and it can be calibrated. I also verified that the accels and gyros still show up.

[image]

dagar, 2021-03-27:
Great, thanks for the quick test @nbort.

commit f6578102 (2021-03-27): boards: mRo x21 enable mpu9250 mag (only internal)

 - fixes #17244
closed 2021-03-27 by dagar