PX4/PX4-Autopilot #17247: Support for multiple SPI drivers of a given type

dakejahl, 2021-03-26:
Support running multiple instances of a driver for a given SPI device, where only the chip selects differ.

I've tried a few different attempts at making this work and I believe the issue is here
https://github.com/PX4/PX4-Autopilot/blob/master/platforms/common/i2c_spi_buses.cpp#L289-L292

I can work around it by doing this before creating the `BusInstanceIterator`
```
cli.type = cli.chipselect_index;
```
but this doesn't guard from instantiating two drivers using the same chip select. 

I think we need to formally support this as a feature!

Thanks!

dagar, 2021-03-27:
There's also the address of the device id here to settle. https://github.com/PX4/PX4-Autopilot/blob/f65781025ebb776fb5c406d96fbbfa52360f8a30/src/lib/drivers/device/nuttx/SPI.cpp#L70-L72

dakejahl, 2021-03-28:
The hacky workaround also prevents interacting with the driver via nsh, like calling "stop". It thinks the driver isn't running because it's `type` doesn't match.

closed 2021-07-08 by dagar