PX4/PX4-Autopilot #17248: EKF2.cpp: Rangefinder instance selection not stable

ecmnet, 2021-03-27:
Having two rangefinders on board (e.g. with HereFlow and a Lightware sensor), the selection of the rangefinder instance depends on the sequence of published UORB messages: [https://github.com/PX4/PX4-Autopilot/blob/f65781025ebb776fb5c406d96fbbfa52360f8a30/src/modules/ekf2/EKF2.cpp#L1557](https://github.com/PX4/PX4-Autopilot/blob/f65781025ebb776fb5c406d96fbbfa52360f8a30/src/modules/ekf2/EKF2.cpp#L1557)
I noticed, that with UAVCAN enabled for HereFlow, the distance sensor of the flow module is selected in EKF2 instead of the Lightware sensor some times. Due to the max_range, the lightwave sensor should be preferred.
My proposal would be to select the rangefinder, that has the highest maximum range instead of the first one found.

dagar, 2021-03-27:
Yes, we need to do something to handle them more explicitly and likely monitor all. The logic exists primarily to filter out non-downward facing options, but then I added a simple timeout to possibly benefit if there's a failure and you have multiple usable options.

ecmnet, 2021-03-29:
Some questions:
- What do you think about the idea using the instance that has the highest max_distance value? 
- I saw that the instance determination is performed only once. Any disadvantages, if the check would be determined continuously?