PX4/PX4-Autopilot #17249: set_gps_global_origin does not enable GPOS in EKF2

ecmnet, 2021-03-27:
Sending a `set_gps_global_origin` message to EKF2 via MAVLINK sets the NED origin as reported by EKF2 but does not 
switch to publish global position. This would be helpful in use cases where no GPS is available and LPOS is provided by other means.

Reason for this is, that the flag `_NED_origin_initialised` is not set to true in`Ekf::setEkfGlobalOrigin`. Doing so in `elf_helper.cpp` would work as long as the parameter `EKF2_DECL_TYPE` is set to 0. Otherwise an invalid look up of the magnetic declination results an invalid EKF2_MAG_DECL value.

ecmnet, 2021-03-30:
Meanwhile, I figured out that setting `_NED_origin_initialised` to true is not appropriate in all cases. I limited that to the case when no map projection is available. This allows to set the origin (and home position) in gps denied situations.

dagar, 2021-03-30:
It might be easier to continue this ECL side. https://github.com/PX4/PX4-ECL/issues/new

Could you try taking a first pass at making the change and we'll work through the remaining details and edge cases?

ecmnet, 2021-04-12:
Addressed with https://github.com/PX4/PX4-ECL/pull/1001

closed 2023-08-26 by dagar