PX4/PX4-Autopilot #17259: No Radio Signal from DSMx on Pixhawk4 mini since V1.11.3

LukasHostettler, 2021-03-28:
Loading the default firmware from qgroundcountrol onto the holybro pixhawk mini results in no signal from the Radio.
The inputs from the RC are not recieved. "Binding Spectrum" however works.

I tested with px4_fmu-v5_default V.1.11.3 (and v1.12.0 Beta 1), both  #versions have the same issue.
The last relased version which worked was V.1.11.2.

Parameters used in all cases:

[20210328_QAV_workingOnV1.11.2.zip](https://github.com/PX4/PX4-Autopilot/files/6217998/20210328_QAV_workingOnV1.11.2.zip)

dagar, 2021-03-28:
Exactly which Spectrum equipment do you have (transmitter and receiver models)?

Are you using PX4 to initiate the bind or doing it externally?

dagar, 2021-03-28:
This is probably the difference impacting you between v1.11.2 and v1.11.3 (and v1.12.0 beta 1). https://github.com/PX4/PX4-Autopilot/pull/16470

LukasHostettler, 2021-03-28:
I use DXE(SPM1000) Transmitter with the SPM 4648 Reciever set to operate with 9 Channels

LukasHostettler, 2021-03-28:
I did only try the binding via px4 since i only have the satellite. The binding seemed to work for two reasons: 

1. The light pattern matched 
2. When downgrading to v.1.11.2 the radio worked immediately

LukasHostettler, 2021-04-03:
Sure, 
the additional information I found on the reciever was:
SPM4648 Contains FCC ID BRWSPMR4648A IC:6157A-SPMR4648A and there is a sticker which says TS904X.
Here are pictures of my setup, what information would you specifically need?

[image]
[image]

The wiring might be a bit more clear on this schema( i left the GPS out) :
[image]

taileron, 2021-04-12:
It is identical on the PixRacer and affects all common Spektrum receivers (SPM 9645,4648,49) that worked without problems until 11.2 and in the master from the beginning of January. 
(its o.k. that 4651 and 4650 cannot work because the SRXL 2 protocol was never supported before.) 
On the Omnibus F4, some DSMX satellites are only recognised (with reduced channel count) after all sticks have been moved or when the arm switch has been activated once.
Up to 11.2 it was immediately there with all channels. 

TX in my case is a DX 10T which can be switched to DX18 compatibility X-Plus 11 and 22ms ... none of these settings make any difference.

taileron, 2021-04-13:
In the previous version (up to 11.2/master 12.2021), channel 11 was also transmitted on 13 and channel 12 on 14, but this was never a reason for me to open an issue because I always need exactly 12 channels. Now the receiver is not recognised at all, which I unfortunately only noticed very late, as most of my test controller setups are without receivers.

taileron, 2021-04-14:
I have now found a small clue. When I connected a 4648 receiver (has only Autobind) to the Pixfalcon, after pressing all the switches once, the RC function suddenly appeared. I noticed that channel 8 is completely ignored.
(It is also still receiving parallel ch 11 on 13 and 12 on 14) 
However, as soon as the value 150% was sent on channel 8, no more RC appears after the boot process.

Very high or low channel values that a transmitter outputs by default should not be able to completely block the detection of the entire RC function!

A late recognition of the receiver also has the consequence that all links between RC channels and parameter tuning no longer function after a restart.

taileron, 2021-04-14:
@LorenzMeier it really seems to be a hard coded limit at 126% as soon as one or more of the TX channels value is above 126%, the whole receiver is no longer detected. If the receiver was detected by chance despite this, the change of the channel that exceeds 126% is ignored separately. 
Where is this parameter hidden to allow channel values up to 150% again which most Spektrum TX offer as highest value like it was up to V11.2?

dagar, 2021-04-14:
It's a protection in the DSM decoder we can lift if you think it's valid. Why do you need to send 150% rather than updating the output side?

dagar, 2021-04-15:
@taileron can you try https://github.com/PX4/PX4-Autopilot/pull/17401?

taileron, 2021-04-15:
Now in all situations on all fc Spektrum rc is detected immediately. 
I think it makes no sense to limit an existing degree of freedom and thus cause an error that is difficult to find. If suddenly no RC receiver is detected at all.
In my planes I usually use several FC connected to a single Spek Diversity Sat, mostly the second one is Betaflight and all flight modes have to switch at exactly the same point. Some switching points are hard coded in Betaflight at 1000 and 2000 uS.
However, what BF interprets as +150% is exactly 2011uS (-150% = 988uS) which corresponds to 2096 or 904uS for PX4.

dagar, 2021-04-15:
> I think it makes no sense to limit an existing degree of freedom and thus cause an error that is difficult to find. If suddenly no RC receiver is detected at all.

It was an attempt to prevent other RC traffic from being incorrect decoded as DSM, even if only for a few samples. The way RC was implemented is it attempts to decode different protocols until it succeeds, then locks on. In retrospect I think I'd prefer it to be more explicitly configured.

dagar, 2021-04-15:
@taileron could you give #17401 one final test so we can merge it?

@LukasHostettler does #17401 change anything for you?

taileron, 2021-04-15:
There are also nice small diversity china style receivers e.g. Speki +, which provide RSSI values at channel outputs and voltage telemetry to the TX and since they also output outside this limit, these receivers are not detected at all or with strange channel number limits.
Spektrum has unfortunately discontinued my favourite receiver with RSSI and telemetry SPM4649 and I am still looking for alternatives, or is SRXL2 (e.g. SPR4650/4651) support planned for the future?

dagar, 2021-04-15:
Thanks for the info @taileron. There are other members of the community looking into SRXL2 support, but I don't know the current status. FYI @teyrana 

I don't currently have any SRXL2 equipment, but the next time I need new equipment I'll likely consider it.

If someone could capture raw serial data for SRXL2 it would probably help getting a working parser together relatively quickly. https://github.com/PX4/PX4-Autopilot/tree/master/test_data

closed 2021-04-15 by LorenzMeier

LukasHostettler, 2021-04-15:
> @LukasHostettler does #17401 change anything for you?

@dagar Is there a way to get at the firmware binaries as artifacts? I'll try on the weekend otherwise.

dagar, 2021-04-15:
It's in master now, but you can always get artifacts for any PR, branch, etc from Github actions.
https://github.com/PX4/PX4-Autopilot/runs/2354802026
Find the NuttX builds, then click on artifacts in the top right. 

[image]

LukasHostettler, 2021-04-15:
Amazing, thanks. Yes, it did fix my problem.

teyrana, 2021-04-16:
@taileron 
"Planned" is a strong word, but I suppose I'm working on it.   
(Currently, I'm in the middle of the px4 learning curve.)