PX4/PX4-Autopilot #17268: Position tracking problem

marcinolokk, 2021-03-30:
Hi all,

We are developing a custom build drone (around 1kg) based on the frame almost identical to s500. We require from it to stay as steady as it is possible in certain moments. Therefore we involved external VIO as source of the position (proved to work well). No GPS involved. However the controller seems not to perform well enough.
I attached the log from the last flight, where the issue appears. Throughout the flight the drone was in position mode and only moved up, hovered for ~30s and landed. There was only movement in Z axis.

The position in x, y, z oscilates ~ 0.3m, 0.15m, 0.2m respectively. This is way too much IMO.
The yaw angle tracking doesn’t perform great as well. The oscilation range is ~7deg. Looking at the Yaw Angular Rate plot it seems that tracking is delayed for ~0.5s however that might be due to the inertia of the body.
Vibrations seem to be a bit high looking at the Actuator Controls FFT and Raw Acceleration (Y axis especially)
I know that tuning PID is the most obvious thing to do, however I already did that and dunno where to look next. Any tips or suggestions would be helpful.

log:
https://review.px4.io/plot_app?log=73a586c8-42bb-4fc2-86f6-6f7d7a8e8c4c

bresch, 2021-03-30:
A few comments that might help you:
- the CG is too far to the front. If possible move the battery a bit back to unload the pitchrate integrator. Otherwise you can change the mixer geometry
- there is a constant yaw control offset; this is often due to a rotor misalignment and creates coupling between the axes, degrading the attitude control performance. You could tilt the rotors by a few degrees (~2-3) using washers under the motors for example to increase yaw authority; this will also improve yaw tracking.
- as you said, the Y accel has large vibrations, this is probably due to the roll rate loop being too sensitive to noise (needs tuning).

ecmnet, 2021-03-30:
Just to give you a benchmark: We achieve with a similar sized frame an accuracy of about 0.02m in XY and 0.04m in Z axis and are still working on it to get below 1cm in all axis. From my experience vibration is one of the crucial factors. I have also seen a big impact to accuracy depending on how close the fc is positioned to the centre of gravity.

marcinolokk, 2021-03-31:
Thanks for the fast reply. 
I corrected 1st and 3th points out of the three you mentioned. I didn't have time and knowledge to look into the the rotor misalignment yet. The vibration on the y axis got lower with smaller pid value. However the oscilations are more less the same. 
Is there a chance that you @ecmnet could share your parameters so I could check what major differences do we have?

ecmnet, 2021-04-02:
Parameters depend heavily on the setup, as with v1.10, I had to increase position control gains even higher than the expected range.

marcinolokk, 2021-04-09:
For the last two days, I played around with the parameters. Reset the gains to default and I found out that EKF2_EV_DELAY was somehow set to 0. Therefore I measured it and set it to 150ms. Even with default PID values, it seems to track the position slightly better (height specifically). I think I also managed to keep the vibrations in a bit smaller range than before.
https://review.px4.io/plot_app?log=eea5efe0-3ac3-4041-bbfb-101d0dd0f049

However new issue appeared. I receive a warning that "RTT is too high for timesync" and "unexpected vision sensor variance" from a console. That appears at around 57s on the attached log. It seems that the estimator "stops listening" to the visual position. The quad starts to move the opposite way as the vision sensor indicates and (my guess: gets out of the vision variance) and switches itself to altitude mode. I was confused that maybe the VIO is diverging so I checked everything with a ground-truth system. Turned out that VIO works well. The attached plots below are from the same flight as the log above, however, placed a little differently in the timeline due to the postprocessing.
[image]
[image]
[image]

Hereby I am confused about what is causing the state estimator to stop following the vision sensor. How the ev_delay param might cause it? Any suggestion on what could we check?? 
Btw, the accuracy of 2 cm in the XY plane sounds impressive.

marcinolokk, 2021-04-15:
This particular issue I solved by enlarging the buffer size using the mentioned trick #9182. Further Improved tracking by setting EKF2_NOISE_MD param. However, I am still confused about dropping the incoming visual position messages.

marcinolokk, 2021-04-27:
Finally, the problem was resolved after updating PX4 Firmware to 1.12 beta3. 
I am not sure how much of the improvement is thanks to the buffer and how much to the hover estimator implemented in the latest release (this certainly improved stability). However, I got a similar tracking performance to @ecmnet.
Thanks!

closed 2021-04-27 by marcinolokk