PX4/PX4-Autopilot #17276: PX4 in SITL mode is saying MAV_CMD_NAV_GUIDED_ENABLE is MAV_RESULT_UNSUPPORTED ?

lovettchris, 2021-03-31:
**Describe the bug**
AirSim uses MAV_CMD_NAV_GUIDED_ENABLE  to request offboard mode, latest master bits are returning the MAV_RESULT::MAV_RESULT_UNSUPPORTED ?  So how does one get the PX4 into offboard mode?  

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a simulator to PX4
2. Wait for it to have good GPS lock and a home position
3. Arm the drone and take off
4. Now try and send MAV_CMD_NAV_GUIDED_ENABLE requesting offboard mode.

**Expected behavior**
Should succeed.

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=cd71c873-321e-4514-8c3c-a3aaca0f459f

**Additional context**
Use this version of AirSim if you are so inclined:
https://github.com/microsoft/AirSim/pull/3546

dagar, 2021-03-31:
I removed this in https://github.com/PX4/PX4-Autopilot/commit/d0c9a5fc93141fd2a393896a3d344db70783e048#diff-72c9e73ba02675201cb800fd25efcb3b3e52e2fb00d22b191ac16f6d886b7c2b because I couldn't find any existing usage, but apparently I didn't search hard enough.

The "normal" usage of OFFBOARD is that you start streaming your offboard setpoints then switch the mode. I can restore `MAV_CMD_NAV_GUIDED_ENABLE` if you think there's value.

lovettchris, 2021-03-31:
I see well I'll try changing AirSim to the new way then, would be nice if there were better docs on this.

dagar, 2021-04-01:
Yes that sounds right.

Roughly what are you using OFFBOARD for? The usage is really geared towards people doing some kind of control via MAVROS and wanting to send setpoints straight into one of the controllers (rate, attitude, velocity, local position). That's why you need to be in OFFBOARD mode then send a continuous stream of setpoints.

If all you want to do is simply move a vehicle around I'd suggest taking a look at MAV_CMD_DO_REPOSITION instead.
https://mavlink.io/en/messages/common.html#MAV_CMD_DO_REPOSITION

lovettchris, 2021-04-01:
AirSim uses offboard for advanced flight control programmed by python scripts mostly using MavLinkSetPositionTargetLocalNed

lovettchris, 2021-04-01:
I got it working now, thanks.

closed 2021-04-01 by lovettchris

dagar, 2021-04-01:
Great, thanks for sharing.

The other thing to note with `OFFBOARD` is that the setpoints are going straight into the controllers. So if you want things like trajectory smoothing you'll need to do it "higher level" going through interfaces like MAV_CMD_DO_REPOSITION that use the same path as mission execution.

Let me know if there's anything we can expand on to improve your use case.

I aspire to one day have a workstation fast enough to start using AirSim. :smirk: