PX4/PX4-Autopilot #17308: Lightware Laser does not start on external bus on Pixhawk4 Fmuv5 with bot sequence

ecmnet, 2021-04-02:
Lightware Laser (Sf10a) does not start with boot sequence on Pixhawk4 (FmuV5). Starting it later via console works without issues:

[image]

[image]

Tested releases:1.11.x,  1.12-beta1 and 1.12-beta2

No issues with external RM3100 magnetometer.

dagar, 2021-04-02:
Do you have `SENS_EN_SF1XX` set?

https://github.com/PX4/PX4-Autopilot/blob/1ec8ce58c7e544e96cbda711d85c7f3b21731b9b/ROMFS/px4fmu_common/init.d/rc.sensors#L68-L72

ecmnet, 2021-04-02:
Yes, of course :D

[image]

dagar, 2021-04-02:
Do you have anything else on that particular bus? Does the manual start ever fail?

I would try setting `_retries = 3` in the constructor. https://github.com/PX4/PX4-Autopilot/blob/1ec8ce58c7e544e96cbda711d85c7f3b21731b9b/src/drivers/distance_sensor/lightware_laser_i2c/lightware_laser_i2c.cpp#L141

ecmnet, 2021-04-02:
Starting manually always works - currently I send the console command via the companion. Never had issues with that approach.

dagar, 2021-04-02:
> Starting manually always works - currently I send the console command via the companion. Never had issues with that approach.

I'm wondering if all the other random I2C probing during startup is causing problems. It's one of the reasons I'd really like the startup of these optional external peripherals to be configured explicitly.

Can you see if the retries help? https://github.com/PX4/PX4-Autopilot/pull/17311

The other thing we can do (that requires slightly more rework) is to add an option that lets the driver start and continue running even if it hasn't been found yet. https://github.com/PX4/PX4-Autopilot/pull/15758

ecmnet, 2021-04-03:
Unfortunately, https://github.com/PX4/PX4-Autopilot/pull/17311 does not solve the issue. Might be related to the HW, as RM3100 is on the same bus and is initialised properly.

dagar, 2021-04-03:
> Unfortunately, #17311 does not solve the issue. Might be related to the HW, as RM3100 is on the same bus and is initialised properly.

Thanks for testing.

Does it make any difference if you disable all the external i2c probing? `SENS_EXT_I2C_PRB 0`
https://github.com/PX4/PX4-Autopilot/blob/588883f6636863c7dfa7278be8eae4d7db330db4/ROMFS/px4fmu_common/init.d/rc.sensors#L112-L147

ecmnet, 2021-04-04:
No difference, if `SENS_EXT_I2C_PRB` is set to 0.

dagar, 2021-04-04:
Does it make a difference if there's nothing else physically on the same bus?

We might need someone with one of these units to step through the code during init to understand what's happening.

ecmnet, 2021-04-05:
I have tried that already of course, no difference.

dagar, 2021-04-05:
Could it be power up timing? Does it take a bit longer for the Lightware to actually start? 

If I had one of these to debug the next thing I'd try is looping (+ small sleep) in the probe.

https://github.com/PX4/PX4-Autopilot/blob/74e848d57ed875d7b03225046ddec49a20693f36/src/drivers/distance_sensor/lightware_laser_i2c/lightware_laser_i2c.cpp#L222-L223

We should implement the keep running (`-k`) option for all of these devices that are explicitly user enabled.

Example https://github.com/PX4/PX4-Autopilot/blob/74e848d57ed875d7b03225046ddec49a20693f36/src/drivers/differential_pressure/sdp3x/SDP3X_main.cpp#L67

ecmnet, 2021-04-08:
I verified that hypothesis using `px4_usleep(200000);`in the probe method of `lightware_laser_i2c.cpp` without success - so I doubt that it is purely a power-up-timing issue.

LorenzMeier, 2021-04-08:
Sleeping 200 ms is not really long. Can you re-test 2 seconds and then start to reduce?

bys1123, 2021-04-11:
PX4FLOW has similar problem #17358

farhangnaderi, 2021-05-13:
I have a similar issue as mentioned here: 

https://discuss.px4.io/t/lightware-lidar-wont-let-the-gps-boot/22334

I was wondering if there is any workaround for it or if any data is needed from me to help debugging this?!

ecmnet, 2021-06-01:
Currently the companion starts the driver using NSH commands:

[image]

jnomikos, 2023-08-23:
I'm seeing this issue on 1.13.2 as well on Flight Core V2

jnomikos, 2023-08-25:
Noticed issue on 1.13.2 FlightCore V2, 1.12.2 FlightCore V2, and 1.13.3 PixHawk4

It appears that the lightware lidar SF11/C does not get detected on any of these boards and attempting to start it shows an error that it is not detected on bus.

The only way I'm able to get it to be detected is manually plug into I2C after the device is already on.

Super weird. Not only that, but if the lightware lidar is already powered on via microusb, it still does not matter, and I run into the same problem.

Since I've noticed this on both the FlightCore V2 and Pixhawk4, it appears to be a bug in PX4. It might explain the problems people are running into here.

jnomikos, 2025-11-20:
> @jnomikos is this still an issue to your knowledge? I do not have an SF11/C to test, but if it's still a problem I can look at the driver

Unsure, haven't tested it since then. Was never resolved on our end