PX4/PX4-Autopilot #17323: SITL tailsitter mission broken

julianoes, 2021-04-05:
**Describe the bug**
It is no longer possible to fly a mission with the SITL tailsitter model. The model oscillate crazily and does not reach waypoints.
[vtol-mission.plan.txt](https://github.com/PX4/PX4-Autopilot/files/6257710/vtol-mission.plan.txt)

**To Reproduce**
Steps to reproduce the behavior:
1. `make px4_sitl gazebo_tailsitter`
2. Uploaded mission plan attached to this issue in QGC.
3. Start mission

**Expected behavior**
It should follow the waypoints and not oscillate and fly away.

`git bisect` points to https://github.com/PX4/PX4-Autopilot/pull/17223

TSC21, 2021-04-05:
One of the problems to solve is this: https://github.com/PX4/PX4-SITL_gazebo/commit/3e5fed04d8e574b10e17e446d2938346bc6152ca#diff-0020558d0aae9f87a3eabf04651b92582eb94089e78c860d631c8a693b523923R90. So we need to make sure that the sensor rotation is correct: https://github.com/PX4/PX4-SITL_gazebo/commit/3e5fed04d8e574b10e17e446d2938346bc6152ca#diff-0020558d0aae9f87a3eabf04651b92582eb94089e78c860d631c8a693b523923L89. But this doesn't seem to solve everything. Even after applying the previous rotation, I cannot get the vehicle to takeoff and I get continuous `ERROR [simulator] poll timeout 0, 22` messages.
@julianoes are you able to fly the tailsitter with a code base previous to the above PR?

Jaeyoung-Lim, 2021-04-05:
@TSC21 You probably need to do a `make clean` in between since the messages need to be regenerated?

TSC21, 2021-04-05:
> @TSC21 You probably need to do a `make clean` in between since the messages need to be regenerated?

Which messages?

julianoes, 2021-04-05:
Will be fixed in https://github.com/PX4/PX4-SITL_gazebo/pull/735.

julianoes, 2021-04-05:
@TSC21 nice catch! We found it in parallel :)

julianoes, 2021-04-05:
> ERROR [simulator] poll timeout 0, 22 messages.

This looks like an unrelated lockstep issue.

TSC21, 2021-04-05:
> > ERROR [simulator] poll timeout 0, 22 messages.
> 
> This looks like an unrelated lockstep issue.

Yep - it was the fact that I had mavlink-router running in the background. Maybe some loopbacks were being introduced or so. All good now.

PR #17324 (2021-04-05): Fix tailsitter SITL airspeed sensor orientation
This fixes the airspeed sensor orientation for tailsitter.

Fixes #17323.
closed 2021-04-05 by TSC21