PX4/PX4-Autopilot #17325: The ellipsoid altitude of global_pos is unstable in SITL

96moustafa, 2021-04-05:
**Describe the bug**
The UTM_GLOBAL_POSITION Mavlink message altitude field gets its data from the ellipsoid altitude of the global vehicle position [here](https://github.com/PX4/PX4-Autopilot/blob/87206556838cefb15ae5517644c7f62c0f4d79dc/src/modules/mavlink/streams/UTM_GLOBAL_POSITION.hpp#L102).

I found that this value is completely unstable in SITL and fluctuates with each message even if the vehicle is not moving.

**To Reproduce**

- Build in simulation: `make px4_sitl_rtps gazebo`
- Use the mavlink inspector to check the alt field of the 
UTM_GLOBAL_POSITION Mavlink message.

**Expected behavior**
The altitude value should be the AMSL altitude.

**Suggesion**
Use `msg.alt = global_pos.alt * 1000.f;` instead of [alt_ellipsoid](https://github.com/PX4/PX4-Autopilot/blob/87206556838cefb15ae5517644c7f62c0f4d79dc/src/modules/mavlink/streams/UTM_GLOBAL_POSITION.hpp#L102)

dagar, 2021-04-05:
Could you share a log showing the problem? According to the mavlink spec that altitude field is supposed to be WGS84. 

http://mavlink.io/en/messages/common.html#UTM_GLOBAL_POSITION

96moustafa, 2021-04-05:
[log_0_2021-4-5-16-25-40.zip](https://github.com/PX4/PX4-Autopilot/files/6258644/log_0_2021-4-5-16-25-40.zip)
I zipped it so it could be uploaded to github. I am not sure if the ellipsoid altitude can be found in the log file.

dagar, 2021-04-05:
Thanks, for future reference you can upload logs to Flight Review (https://logs.px4.io/). https://logs.px4.io/plot_app?log=65441f2f-c3ec-40c9-92b5-0c0c36eb21ad

It looks like GPS `alt_ellipsoid` isn't being populated in SITL. I'll make a quick fix.

dagar, 2021-04-05:
@96moustafa can you try https://github.com/PX4/PX4-Autopilot/pull/17330?

96moustafa, 2021-04-05:
> @96moustafa can you try #17330?

Works!

closed 2021-04-06 by dagar