PX4/PX4-Autopilot #17328: Couls not get GPS status from Telem2, firmware master but v1.11.3 OK

saengphet, 2021-04-05:
**Describe the bug**
I set up dual GPS, primary GPS is at GPS1 port and secondary is at telem2 port. Pixhawk 4 with Firmware version v1.11.3 works fine. However, after I upgraded firmware to master version. I could not get signal from secondary GPS (using gps status command). After that, I changed serial port from Telem2 to Serial4 to connect GPS, it works as expected !!. I have tested these with different fcu and gps. The result is the same.

**To Reproduce**
Steps to reproduce the behavior:
1. Set GPS1 (or GPS2) to accept signal from telem2, set serial baud-rate to auto.
2. Restart the fcu
3. Run "gps status" at console
4. Could not set gps signal
5. From 1. Change to accept gps signal from telem2 to serial 4 (physical port is uart & i2c B), then you can get the gps signal

**Expected behavior**
Can get the GPS signal from either telem2 or serial 4

**Drone (please complete the following information):**
- Quadrotor 
- pixhawk 4
- pixhawk 4 GPS
- Drotek Sirius GPS connect to telem2 (or serial4) of pixhawk. Also test using Drotek F9P rover gps.

dagar, 2021-04-05:
Do you have a flight log from master showing the problem? https://logs.px4.io/

In particular I'm looking for the specific values of the GPS* parameters and any dmesg output.

mcsauder, 2021-04-06:
Hi @saengphet ,

In re-wiring a holybro GPS JST-GH housing to see if I could reproduce your issue I noticed that perhaps, (but perhaps not), the reason your are encountering success is the pinout/GPIO configuration difference between the two ports you are attempting to utilize and what the hardware expects on pins 4/5 of telem1/2 vs UART&I2CB pins 4/5 and the Serius A2/A3.  Allow me to illustrate:
[image]

Pixhawk 4 pinouts;
[image]

I don't have a Drotek Serius to test with, but I can at least state that I can replicate your results with the Holybro GPS.

-Mark

saengphet, 2021-04-06:
> Do you have a flight log from master showing the problem? https://logs.px4.io/
> 
> In particular I'm looking for the specific values of the GPS* parameters and any dmesg output.

pixhawk4+F9P:serial4 - good
https://review.px4.io/plot_app?log=1ff8d5b2-fc33-442f-a602-c74968ac38ca

pixhawk4+F9P:serial4 - bad
https://review.px4.io/plot_app?log=6b6bdee0-9c31-4220-a54b-4e7ed6ed44b3

saengphet, 2021-04-06:
> Hi @saengphet ,
>
> In re-wiring a holybro GPS JST-GH housing to see if I could reproduce your issue I noticed that perhaps, (but perhaps not), the reason your are encountering success is the [...]

Hi @mcsauder, thank you for your reply. I'm not sure that my problem relates to what you mention or not. In my case, If I flash firmware v1.11.3, I can use either telem2 or serial4 port "without re-wiring", however after I upgraded the firmware to master, only serial4 works fine but telem2 not. In my point of view, it's not a hardware bug but the firmware does it.

mcsauder, 2021-04-06:
Hi @saengphet , my thought was that with recent upgrades at the hardware level in NuttX, the CTS/RTS lines might now be being held high/low properly or improperly where they were not previously and it might be hanging the GPS hardware in a strange state.  I think that is what is going on with the holybro GPS between the releases.

saengphet, 2021-04-06:
> Hi @saengphet , my thought was that with recent upgrades at the hardware level in NuttX, the CTS/RTS lines might now be being held high/low properly or improperly where they were not previously and [...]

I will try to disconnect I2C pin and connect to telem2 port and show you the result. Thanks

Igor-Misic, 2021-04-07:
We also have problems with using GPS at **TELEM** ports at **FMUv5x**. 
If we use TELEM1 then we are randomly losing connection to GPS. 
It can be observed as GPS status going from **OK** to **NOT OK**
```
nsh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] status: OK, port: /dev/ttyS6, baudrate: 115200
INFO  [gps] sat info: disabled

nsh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] status: NOT OK, port: /dev/ttyS6, baudrate: 115200
INFO  [gps] sat info: disabled
INFO  [gps] rate reading:             93 B/s
INFO  [gps]
```

If we change it from TELEM1 to TELEM3 then this doesn't happen anymore.

@davids5 does any of this look familiar to you so you can help us here? (any changes with CST and RTS at Nuttx level)

davids5, 2021-04-07:
The HS has not changed in NuttX. Both port have RXDMA. The driver uses the same code on both ports.  

```
TELEM1 is UART7

CONFIG_UART7_BAUD=57600
CONFIG_UART7_IFLOWCONTROL=y
CONFIG_UART7_OFLOWCONTROL=y
CONFIG_UART7_RXBUFSIZE=600
CONFIG_UART7_RXDMA=y
CONFIG_UART7_TXBUFSIZE=3000

TELEM2 UART5

CONFIG_UART5_IFLOWCONTROL=y
CONFIG_UART5_OFLOWCONTROL=y
CONFIG_UART5_RXBUFSIZE=600
CONFIG_UART5_RXDMA=y
CONFIG_UART5_TXBUFSIZE=3000
CONFIG_UART5_TXDMA=y
```

What does mavlink status show?

Please re-test after doing a mavlink stop-all

Igor-Misic, 2021-04-07:
I withdraw my problem from here. It seems it is different nature and it also happening at TELEM3.

mrpollo, 2021-04-28:
Hey @saengphet  are you still experiencing this issue with the latest stable or beta releases?

saengphet, 2021-04-29:
> Hey @saengphet  are you still experiencing this issue with the latest stable or beta releases?

Hi, I will test it again ASAP. For now, I use the Serial 4 (UART) instead.