PX4/PX4-Autopilot #17337: Improve Robustness of `LANDING_TARGET` Message

obicons, 2021-04-05:
**Describe the bug**
In `MavlinkReceiver::handle_message_landing_target`, if `landing_target.frame != MAV_FRAME_LOCAL_NED` then the message is handled as a "Target Relative to Captured Image" message. [Per MAVLink's documentation],(https://mavlink.io/en/services/landing_target.html) PX4 does not handle frames other than `MAV_FRAME_LOCAL_NED`. However, treating messages with other frames as a target relative to a captured image does not seem correct. MAVLink appears to be silent on the issue of default parameter values for this message. Users may supply values of `angle_x`, etc that cause undesirable behavior.

**To Reproduce**
N/A

**Expected behavior**
In the best case, if `position_valid` is `1`, and the value of `frame` is not `MAV_FRAME_LOCAL_NED`, I would like feedback advising me that the coordinate frame is not supported. In the worst case, I expect my MAVLink message to be ignored.

**Log Files and Screenshots**
N/A

Add screenshots to help explain your problem.
N/A

**Drone (please complete the following information):**
N/A

**Additional context**
N/A

dagar, 2021-04-05:
Could you open a PR adding the desired behavior? 

https://github.com/PX4/PX4-Autopilot/blob/54438140e12b44654b08e890399f52d564d89874/src/modules/mavlink/mavlink_receiver.cpp#L2289

You can simply only support (republish) the specific frames that work, then anything else can be rejected with a mavlink critical message like this. https://github.com/PX4/PX4-Autopilot/blob/54438140e12b44654b08e890399f52d564d89874/src/modules/mavlink/mavlink_receiver.cpp#L960-L961

obicons, 2021-04-05:
Sure, thank you for the fast response!

obicons, 2021-04-05:
@dagar There is similar logic in `void Simulator::handle_message_landing_target(const mavlink_message_t *msg)`. Should this be updated as well?

dagar, 2021-04-05:
> @dagar There is similar logic in `void Simulator::handle_message_landing_target(const mavlink_message_t *msg)`. Should this be updated as well?

Yes, and instead of the `else` accepting everything you could limit it to only what's actually supported.

obicons, 2022-01-01:
This should be closed, since the PR was merged.

closed 2022-01-01 by obicons