PX4/PX4-Autopilot #17339: Found three parameters problems in v1.12.beat2

Kim-zhi-jiang, 2021-04-06:
Hello there

Found three parameters problems in v1.12.beat2

ERROR [param] Parameter FW_RSP_OFF not found.
But in here have: "PARAM_DEFINE_FLOAT(FW_PSP_OFF, 0.0f);"
/PX4-Autopilot/src/modules/fw_att_control/fw_att_control_params.c

ERROR [param] Parameter PWM_DISARMED not found.
Deleted this file in v1.12.beat2 "/PX4-Autopilot/src/modules/sensors/pwm_params_main.c"
I don’t know add where in the new planning.

ERROR [param] Parameter PWM_AUX_DISARMED not found.
Deleted this file in v1.12.beat2  "/PX4-Autopilot/src/modules/sensors/pwm_params_main.c"
I don’t know add where in the new planning.

dagar, 2021-04-06:
- FW_RSP_OFF
    - removed because I've never seen anyone actually need it
 - PWM_DISARMED
    - now PWM_MAIN_DISARM (because it applies to only the MAIN output group)
    - user set PWM_DISARMED is automatically transitioned to PWM_MAIN_DISARM
 - PWM_AUX_DISARMED
    - now PWM_AUX_DISARM for naming consistency

Kim-zhi-jiang, 2021-04-06:
@dagar 

Now I see

Thank you very much for replying!

closed 2021-04-06 by Kim-zhi-jiang