PX4/PX4-Autopilot #17347: Documentation of attitude quaternion direction

manumerous, 2021-04-06:
Hi 

The repository https://github.com/PX4/Devguide is archived and It is not possible to open an Issue there as suggested. I therefore hope that it is okay to open the Issue here. 

I would like to know the exact details of what the attitude quaternions, as found in the ulog under for example `vehicle_attitude`, describe. In particular which two frames it relates and in which direction (e.g. if the quat is transformed to a rotation matrix does that matrix convert A to B or B to A). 
I guess a logical candidate would be to relate the `MAV_FRAME_BODY_NED` to `MAV_FRAME_BODY_FRD`. However, I could not find any documentation on that topic. 

Any help is appreciated.

dagar, 2021-04-06:
It's all been merged into the user guide. https://github.com/PX4/PX4-user_guide

manumerous, 2021-04-06:
@dagar Thank you for the link. I also could not find any detailed documentation about what PX4s quaternions represent there. Would you propose to close this issue here and post the question again on that repository? 

Maybe it would be possible to change the description when opening a documentation issue to point to the new repo? Currently it stilllinks https://github.com/PX4/Devguide.

closed 2025-11-06 by hamishwillee