PX4/PX4-Autopilot #17352: Cube Orange USB Serial not working on WIndows

dlech, 2021-04-06:
**Describe the bug**

Cube Orange flashed with PX4 firmware does not connect on Windows due to the serial port driver not loading for the mavlink connection.

**To Reproduce**
Steps to reproduce the behavior:

1. Flash PX4 firmware using QGroundControl (using developer/master)
  [image]
2. After the autopilot reboots, it does not connect
3. Go to *Device Manager* and notice an error
  [image]
4. Double-click it to see the details
  [image]

**Expected behavior**

Cube Orange should be able to connect the same as other autopilots. 

**Additional context**

The reason I think this is a firmware issue rather than an issue with my computer is that other autopilots work (Cube Black and original Pixhawk) and if I flash the ArduPilot firmware instead of the PX4 firmware to the Cube Orange, I don't get this error.

[image]

[image]

saengphet, 2021-04-06:
I am not sure this can solve your problem or not but it works for me. I installed the latest version of mission planner, then try to connect cube orange again in QGC. In my opinion it is about cube orange usb driver in windows, in Linux works fine without above installation.

dlech, 2021-04-06:
Thanks for the tip! I can confirm that installing the latest mission planner fixed the issue for me as well.

saengphet, 2021-04-06:
> Thanks for the tip! I can confirm that installing the latest mission planner (v1.3.74) fixed the issue for me as well.

Wonderful !! However, we still need the right solution for other users. @dagar

mrpollo, 2021-04-28:
@davids5 Can you please give this a quick test? it should just work with our bootloader on windows.

MaEtUgR, 2021-04-28:
@davids5 was just able to reproduce this. We're not sure what the bootloader does to get into this situation.

dlech, 2021-04-28:
>           ===>Device Descriptor<===
> bLength:                           0x12
> bDescriptorType:                   0x01
> bcdUSB:                          0x0200
> bDeviceClass:                      0x02  -> This is a Communication Device
> bDeviceSubClass:                   0x00
> bDeviceProtocol:                   0x00
> bMaxPacketSize0:                   0x40 = (64) Bytes
> idVendor:                        0x2DAE = Vendor ID not listed with USB.org as of 02-15-2012
> idProduct:                       0x1016
> bcdDevice:                       0x0101
> iManufacturer:                     0x01
>      English (United States)  "CubePilot"
> iProduct:                          0x02
>      English (United States)  "PX4 BL CubePilot CubeOrange"
> iSerialNumber:                     0x03
>      English (United States)  "0"
> bNumConfigurations:                0x01

Should `bDeviceSubClass` be 0x02 to match the built-in Windows USB serial driver? https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/usb-driver-installation-based-on-compatible-ids#windows10

n-snyder, 2021-08-19:
I am having the same issue from Windows 10 using either Mission Planner 1.3.75 or today's daily build of QGC. I can install ArduPilot fine, and communicate with it afterward. However, when installing PX4 v4.1.3 from either MP or QGC, I can install PX4 fine, but the USB comm port for Mavlink is invalid.

davids5, 2021-08-20:
My Windows 10 machine is down at the moment. But I would guess that the driver install in QGC is not up to date with the mission planers copy. We should see if a virgin install of QGC has the inf files with 
a VID/PID
idVendor:                        0x2DAE 
idProduct:                       0x1016

If not that is the root cause.

@mrpollo - remember the test we did on you Win10 machine. It is the same that is needed here.

trddo, 2022-11-21:
Was there a final conclusion, (fix) for this issue?
I am also experiencing this same phenomenon.
I was temporarily able to connect after uninstalling all drivers, then reinstalling latest drivers.msi PX4, however this only worked for a one time connection.

mnumanuyar, 2022-12-13:
>  But I would guess that the driver install in QGC is not up to date with the mission planers copy. 

I had this problem, with device manager and all. installing mission planner solved it for me

julianoes, 2023-05-07:
I can reproduce some of the problems. I'm having a look.

julianoes, 2023-05-07:
As @davids5 pointed out above. The likely culprit is that [QGC's driver.msi is from 2020](https://github.com/mavlink/qgroundcontrol/commit/513ab1ed252f8b0bb7e3d99fc9ffca9b7ee035ae) and needs to be updated.

I'm trying to update it: https://github.com/mavlink/qgroundcontrol/pull/10679

dlech, 2023-05-07:
It would be even better if the firmware was modified to included the required USB descriptors so that drivers are not required at all on Windows.

See https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/supported-usb-classes

closed 2023-05-08 by mrpollo

julianoes, 2023-05-08:
@dlech ok, how/what would we have to change?

dlech, 2023-05-08:
Since this is a serial port, I think just changing the device subclass to 0x02 should do it.

https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/usb-driver-installation-based-on-compatible-ids#windows10-and-later

julianoes, 2023-05-08:
Where? Sorry I'm not sure how this all works.

dlech, 2023-05-08:
I did some digging. It looks like USB is handled in NuttX.

Based on the descriptor listings in previous comments, it looks like this device descriptor is being used which has subclass of `CDC_SUBCLASS_NONE` which is 0x00. See https://github.com/PX4/NuttX/blob/eb43c582eacff396c49a878b6e067ef66b8841d3/drivers/usbdev/cdcacm_desc.c#L49-L68

There are also interface association descriptors that have subclass of `CDC_SUBCLASS_ACM` which is the 0x02 that Windows uses for driver matching. See https://github.com/PX4/NuttX/blob/eb43c582eacff396c49a878b6e067ef66b8841d3/drivers/usbdev/cdcacm_desc.c#L412-L447

So, I think it is probably a matter of changing the NuttX config to make the USB device appear as a composite device instead of strictly as a communication device then Windows will use the interface associate descriptors to figure out which driver to use.

dlech, 2023-05-08:
Or perhaps more simply, change this line to `CDC_SUBCLASS_ACM`.

https://github.com/PX4/NuttX/blob/eb43c582eacff396c49a878b6e067ef66b8841d3/drivers/usbdev/cdcacm_desc.c#L66

julianoes, 2023-05-08:
Interesting. @davids5 would that make sense?

MaEtUgR, 2023-05-09:
But is it a cube thing then? I currently don't have any cube board at hand but e.g. pixhawk 4 (+ mini), pixracer just connect to the ground station and can be flashed under Windows 10/11 without installing any driver. QGC at installation always prompts you to install additional drivers, I just press abort and it all works for years now already.

davids5, 2023-05-09:
> Interesting. @davids5 would that make sense?

I am not sure. The notes say you need a inf file for 8.1 So we have always had one. If windows 10+ qualifies the driver on the sub-class it may work. I am having troubles finding the use of "sub-class" So I would try the suggested change and test it.