PX4/PX4-Autopilot #17364: dshot 3d support missing

benjinne, 2021-04-08:
**Describe the bug**
The dshot output does not work correctly when using blheli with 3d mode enabled. For example when using a blheli ESC for a rover mixer that  supports a bi-directional esc, only pwm works currently. I think this is due to the bottom half of dshot commands being reversed.

I've found several explanations on how this works, and it just needs implemented to work correctly.
Copied from here: https://www.swallenhardware.io/battlebots/2019/4/20/a-developers-guide-to-dshot-escs
```
DShot Bidirectional Motion (3D Mode)
If you enable 3D mode, the throttle ranges split in two.

Direction 1) 48 is the slowest, 1047 is the fastest

Direction 2) 1049 is the slowest, 2047 is the fastest

1048 does NOT stop the motor! Use command 0 for that.
```
This is working in betaflight.
Wiki documentation: https://github.com/betaflight/betaflight/wiki/3D-Setup
Small explanation in the betaflight code: https://github.com/betaflight/betaflight/blob/025ee87a7aca068e3659fd066b8a9afbed123361/src/main/drivers/dshot_command.h#L28 

**Suggested Behavior**
1. Add a DSHOT_3D boolean parameter that changes the dshot behavior as described above
2. Add deadband parameters as described in the betaflight wiki

**Potential Use Cases**
1. My use case is for a rover with 4 independent motors (no servos for steering)
2. The typical rover with a PWM servo for steering and a dshot esc ( I'm not sure how the mixing would work)
3. Copters can thrust downwards without flipping upside down or fly upside down

benjinne, 2021-04-08:
This is related to [Support normal car ESC with reverse](#11992)

bkueng, 2021-04-09:
You're right, this is currently not implementent, and the bottom half needs to be inverted. Can you do a PR with your suggesion?

benjinne, 2021-04-14:
> You're right, this is currently not implementent, and the bottom half needs to be inverted. Can you do a PR with your suggesion?

I would start working on this, but my use case is for a rover, and I don't have Dshot working at all on a Rover. See #17381

PR #17488 (2021-05-06): Add 3D DShot capability
**Describe problem solved by this pull request**
Fixes #17364 

**Describe your solution**
Added 3 DShot parameters to enable 3D DShot.
If DSHOT_3D_ENABLE is selected, the bottom actuator_output range is reversed to comply with DShot specification.
DSHOT_3D_DEAD_H and DSHOT_3D_DEAD_L set the range where output is 0

**Describe possible alternatives**
- Use generic deadband parameters that apply to both DShot and pwm
- Fix actuator output value of 0 not being usable by DShot [see here](https://github.com/benjinne/PX4-Autopilot/blob/4f52c0b6daa93f01b3c02e9457265be107af74b [... cut]
closed 2021-05-06 by bkueng