PX4/PX4-Autopilot #17366: Vehicle Setup>Power Setup

mommotti, 2021-04-09:
I'm trying to set up my Pixhawk 2.1 using PX4.
I don't have a battery, the Pixhawk is connected to my companion computer ( Raspberry Pi 3b ) via USB.
I have already disabled the USB checks with " CBRK_USB_CHK 197848 "
Is there a way to bypass the Power Setup and just use the USB connection provided by the Raspberry to power the Pixhawk?
Power Setup:
[image]
Parameters (Battery Calibration): 
[image]
The build:
[image]
Thank you in advance!

benjinne, 2021-04-10:
Try setting CBRK_SUPPLY_CHK to arm with no power input

mommotti, 2021-04-11:
> 
> 
> Try setting CBRK_SUPPLY_CHK to arm with no power input

[image]
Thanks for the quick reply! Still nothing tho... The problem persists

LorenzMeier, 2021-04-11:
What is the error message you are getting?

mommotti, 2021-04-12:
> 
> 
> What is the error message you are getting?

I was getting something similar to "Unable to arm, please complete the setup."
But now that warning message doesn't pop up anymore on QGC... Thank you all, let's call it close for now

closed 2021-04-12 by mommotti