PX4/PX4-Autopilot #17369: Safety Button can't back to lock status on Pixhawk4

bys1123, 2021-04-11:
After flash current master, Safety Button can't back to lock status on Pixhawk4

saengphet, 2021-04-11:
> After flash current master, Safety Button can't back to lock status on Pixhawk4

Hi, @bys1123 I have tested with cube orange and got the same result. After press the safety switch button once to pre-arm on state, it could not turn to pre-arm off state, the LED still be "double blink" instead of slow blink.

LorenzMeier, 2021-04-11:
This is by design (see the release notes) because we have seen accidental in-air triggering of the safety switch due to poor wiring. So now it is latching.

bys1123, 2021-04-11:
@LorenzMeier Thanks for explain, but in disarm status, i think safety button is still useful, especially in research environment. In arm status, I agree make safety switch un-tigger-able.

ThomasRigi, 2021-04-29:
@LorenzMeier Are there any plans to fix this for v1.12? I agree that latching is good for when the drone is armed, but it has be possible to un-prearm when the drone is disarmed as @bys1123 is saying.

ThomasRigi, 2021-04-29:
And could you please send the reference to the PR that changed to latching behaviour? It's not linked in the release notes

LorenzMeier, 2021-04-29:
I will look into it. My main concern is to keep it consistent and I need to see if that's consistently doable before merging the PX4 IO rework (which will definitely allow for it).

dlech, 2021-09-24:
> And could you please send the reference to the PR that changed to latching behaviour? It's not linked in the release notes

I searched for related commits:
- https://github.com/PX4/PX4-Autopilot/commit/867b006861d4ea887d1f17a79d84b4455a392c91
- https://github.com/PX4/PX4-Autopilot/commit/7157584fe11dbcf644928f877964d0c2ff7dad90
- https://github.com/PX4/PX4-Autopilot/pull/16751

spurnvoj, 2022-03-16:
Hi, any update on this?

ThomasRigi, 2022-05-06:
@LorenzMeier @MaEtUgR @dagar can we get this fixed ?

dagar, 2022-05-06:
> @LorenzMeier @MaEtUgR @dagar can we get this fixed ?

@ThomasRigi yes, maybe we can talk about it in https://github.com/PX4/PX4-Autopilot/pull/19558 and review your specific requirements?

The original change to have the px4io safety button latch was to remove the possibility of accidental disarm in air. Additionally we also changed the safety button to disabled by default because that's what the vast majority of people were doing and it doesn't really for so many systems that use more than just the px4io outputs.

Anyway, sorry if this disrupted anything for you, let's review the core requirements?

ThomasRigi, 2022-05-13:
Ok, I commented in #19558. :+1: