PX4/PX4-Autopilot #17381: Dshot on rover not working

benjinne, 2021-04-12:
**Describe the bug**
On 1.11.3, When I select Dshot for my rover, my blheli ESCs refuse to initialize, however quad x works and gives the final beeps on power up. It appears the dshot zero command isn't being sent correctly or at all. 

**Expected behavior**
Rover airframes should be able to run Dshot 

**Additional context**
I suspected the rc.rover_defaults lowering the PWM_RATE to be the problem, but commenting that out doesn't fix the problem. I'm not sure what else could be preventing Dshot from operating correctly. I'll attach an oscilloscope soon to see if it's a timing issue or if there's no output at all.

bkueng, 2021-04-15:
What is the output of `dshot status`? And `uorb top actuator`?
Maybe the update rate is too low.

benjinne, 2021-04-19:
```
NuttShell (NSH)
nsh> dshot status
INFO  [dshot] Mode: outputs8
INFO  [dshot] Outputs initialized: yes
INFO  [dshot] Outputs on: yes
dshot: cycle: 4381 events, 118571us elapsed, 27.06us avg, min 2us max 306us 4.985us rms
control latency: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
INFO  [mixer_module] Switched to rate_ctrl work queue: 1
INFO  [mixer_module] Mixer loaded: yes
INFO  [mixer_module] Driver instance: 0
INFO  [mixer_module] Channel Configuration:
INFO  [mixer_module] Channel 0: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
INFO  [mixer_module] Channel 1: value: 1054, failsafe: 0, disarmed: 0, min: 109, max: 1999
INFO  [mixer_module] Channel 2: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
INFO  [mixer_module] Channel 3: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
INFO  [mixer_module] Channel 4: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
INFO  [mixer_module] Channel 5: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
INFO  [mixer_module] Channel 6: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
INFO  [mixer_module] Channel 7: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
```
and
```
nsh> uorb top actuator

[0Jupdate: 1s, num topics: 70
TOPIC NAME        INST #SUB RATE #Q SIZE
actuator_armed       0    7    1  1   24 
actuator_controls_0  0    4  197  1   48 
actuator_outputs     0    5  197  1   80 
```

benjinne, 2021-04-19:
Dshot on quad x airframe

```
nsh> dshot status
INFO  [dshot] Mode: outputs8
INFO  [dshot] Outputs initialized: yes
INFO  [dshot] Outputs on: yes
dshot: cycle: 103050 events, 5044880us elapsed, 48.96us avg, min 2us max 413us 3.905us rms
control latency: 103048 events, 30634013us elapsed, 297.28us avg, min 282us max 855us 8.085us rms
INFO  [mixer_module] Switched to rate_ctrl work queue: 1
INFO  [mixer_module] Mixer loaded: yes
INFO  [mixer_module] Driver instance: 0
INFO  [mixer_module] Channel Configuration:
INFO  [mixer_module] Channel 0: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
INFO  [mixer_module] Channel 1: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
INFO  [mixer_module] Channel 2: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
INFO  [mixer_module] Channel 3: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
INFO  [mixer_module] Channel 4: value: 1054, failsafe: 0, disarmed: 0, min: 109, max: 1999
INFO  [mixer_module] Channel 5: value: 1054, failsafe: 0, disarmed: 0, min: 109, max: 1999
INFO  [mixer_module] Channel 6: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
INFO  [mixer_module] Channel 7: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
nsh> uorb top actuator

[0Jupdate: 1s, num topics: 74
TOPIC NAME        INST #SUB RATE #Q SIZE
actuator_armed       0    7    2  1   24 
actuator_controls_0  0    5  787  1   48 
actuator_controls_3  0    1    0  1   48 
actuator_outputs     0    5  787  1   80 
[0Jnsh> 
```

benjinne, 2021-04-19:
Just tested with a custom rover mixer that outputs dshot 0 packets instead of middle value and I have the same issue with esc's not initializing. Hooking up my oscilloscope shows both quad and rover have a packet length of around 100us for dshot150 which is expected, however the length between packets varies greatly. The quad has around a 1ms delay between packets, but the rover has about 4.5ms delay between packets. I think if the delay is long enough between packets, the esc will never initialize or disarm.

bkueng, 2021-04-20:
> `actuator_controls_0  0    4  197  1   48 `

The outputs are only updated with 200Hz for rover, and I've seen some ESC's refusing to initialize with a too low update rate.
The rate comes from the local postion update in https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/rover_pos_control/RoverPositionControl.cpp#L401, and you could test with publishing `actuator_controls_0` with the rate of the gyro.

benjinne, 2021-04-20:
After I switched the RoverPositionControl to update on vehicle_angular_velocity as the MCRateControl does, I still get an update rate of around 200hz. I then increased `IMU_GYRO_RATEMAX` to 800 as it's the default for multicopters, however that still did not increase the actuator frequency.

benjinne, 2021-04-22:
@bkueng I found a solution! My problem was I needed to change the callback function to use the `_vehicle_angular_velocity_sub` instead as the MCRateControl does.

I just tested on a blheli_s esc and found it stopped initializing at around 10Hz which would have no problem with the current code, however the ESC on my rover doesn't seem to initialize with the current 200Hz actuator output for some reason. Perhaps because it is blheli32. Regardless the PR I submitted defaults the 400Hz IMU_GYRO_RATEMAX parameter which works.

PR #17445 (2021-04-26): Rover publish actuator controls with rate of gyro
**Describe problem solved by this pull request**
Fixes #17381 
Fix Dshot not initializing on certain ESCs requiring update rate faster than 250Hz, by making the actuator update configurable by adjusting the IMU_GYRO_RATEMAX parameter. 

**Describe your solution**
Make Rover update on `vehicle_angular_velocity` instead of `vehicle_attitude`

**Describe possible alternatives**
Increase the rate of `vehicle_attitude`

**Test data / coverage**
Tested on the modalai/fc_v1 board by selecting rover and measuring update rate with oscilloscope and `uorb [... cut]
closed 2021-04-26 by bkueng