PX4/PX4-Autopilot #17384: Unexpected switching from Position to Altitude with strong GPS

tylervandenberg, 2021-04-13:
**Describe the bug**
Using a MRO Dual N8N GPS,
(https://store.mrobotics.io/mRo-GPS-u-Blox-Neo-M8N-Dual-Compass-LIS3MDL-IST831-p/m10034-li8303.htm)
we have had vehicles on multiple occasions switch to altitude mode mid-flight unexpectedly. The latest event resulted in a vehicle tipping over upon landing. In hindsight we believe it may have been possible to manually switch back to position mode before landing, but can not confirm. We are also using a ModalAI Flight Core. 

**To Reproduce**
Steps to reproduce the behavior:
1. Drone powered on
2. Took off in position flight mode
3. Unexpectedly switched to altitude flight mode shortly after
4. See error

**Expected behavior**
Since the vehicle accepted compass calibration and was able to arm, expected flight would have been a complete flight in Position mode.

**Log Files and Screenshots**
Flight Review:
(https://review.px4.io/plot_app?log=6f68bf47-590b-4c7d-8c2f-b5815c23b003)

It can be seen from the following screenshot that the number of satellites used was always above 15, and the GPS fix and positional accuracy seemed normal. 
[image]

It can also be seen from the following screenshot that several error messages were logged around the time of the switch.
[image]

**Drone:**
IF1200, ModalAI Flight Core, MRO Dual N8N GPS

LorenzMeier, 2021-04-13:
You have excessive vibration on the airframe that would need to be addressed.

closed 2021-04-13 by LorenzMeier

tylervandenberg, 2021-04-13:
@LorenzMeier I should point out that the vehicle tipped over around 3 minutes and 6 seconds, and that is when the excessive vibration starts. There was very high wind and the ground was sloped, making it difficult to land in altitude mode. Please reopen this issue and only consider the start of the flight through 3 minutes.

evigdorchik, 2021-04-13:
@tylervandenberg I have seen this issue as well and don't have a solution either. Here are several issues that seem related:

https://discuss.px4.io/t/no-local-position/10604 (unsolved)
https://discuss.px4.io/t/failsafe-enabled-no-local-position/11210 (unsolved)

The status flag changed unexpectedly in state_machine_helper.cpp but I can't trace why.

This seems to be an issue in various other systems over the past few years. @LorenzMeier can you suggest what fixed the above issued, suggest where else to start a discussion, or reopen this issue? I am also interested in finding a solution.