PX4/PX4-Autopilot #17396: Unable to build and run jMAVsim

rajeshroy402, 2021-04-14:
Hello, 
I followed the command which was mentioned [here](https://docs.px4.io/master/en/dev_setup/dev_env_linux_ubuntu.html) for Ubuntu build but when I run `make px4_sitl jmavsim` or `make px4_sitl_defautl jmavsim`I get error. 

Kindly check the last few lines of my terminal print with the error:
```
FAILED: generated_params/serial_params.c 
cd /home/rajesh/Developer/PX4-Autopilot/build/px4_sitl_default/src/lib/parameters && /usr/bin/cmake -E make_directory /home/rajesh/Developer/PX4-Autopilot/build/px4_sitl_default/generated_params && /usr/bin/python3 /home/rajesh/Developer/PX4-Autopilot/Tools/serial/generate_config.py --params-file /home/rajesh/Developer/PX4-Autopilot/build/px4_sitl_default/generated_params/serial_params.c --serial-ports --ethernet --config-files /home/rajesh/Developer/PX4-Autopilot/src/lib/battery/module.yaml /home/rajesh/Developer/PX4-Autopilot/src/lib/pwm/pwm_main_params.yaml /home/rajesh/Developer/PX4-Autopilot/src/lib/pwm/pwm_aux_params.yaml /home/rajesh/Developer/PX4-Autopilot/src/lib/pwm/pwm_extra_params.yaml /home/rajesh/Developer/PX4-Autopilot/src/drivers/gps/module.yaml /home/rajesh/Developer/PX4-Autopilot/src/modules/mavlink/module.yaml
Failed to import jinja2: No module named 'jinja2'

You may need to install it using:
    pip3 install --user jinja2

ninja: build stopped: subcommand failed.
make: *** [Makefile:221: px4_sitl_default] Error 1
```

rajeshroy402, 2021-04-14:
I want to highlight that when I try to execute 'bash ./PX4-Autopilot/Tools/setup/ubuntu.sh' , it is unable to install Python3 dependencies.

```
Installing PX4 Python3 dependencies
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(136, '[X509: NO_CERTIFICATE_OR_CRL_FOUND] no certificate or crl found (_ssl.c:4263)'))': /simple/argcomplete/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(136, '[X509: NO_CERTIFICATE_OR_CRL_FOUND] no certificate or crl found (_ssl.c:4263)'))': /simple/argcomplete/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(136, '[X509: NO_CERTIFICATE_OR_CRL_FOUND] no certificate or crl found (_ssl.c:4263)'))': /simple/argcomplete/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(136, '[X509: NO_CERTIFICATE_OR_CRL_FOUND] no certificate or crl found (_ssl.c:4263)'))': /simple/argcomplete/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(136, '[X509: NO_CERTIFICATE_OR_CRL_FOUND] no certificate or crl found (_ssl.c:4263)'))': /simple/argcomplete/
Could not fetch URL https://pypi.org/simple/argcomplete/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/argcomplete/ (Caused by SSLError(SSLError(136, '[X509: NO_CERTIFICATE_OR_CRL_FOUND] no certificate or crl found (_ssl.c:4263)'))) - skipping
ERROR: Could not find a version that satisfies the requirement argcomplete (from -r /home/rajesh/Developer/PX4-Autopilot/Tools/setup/requirements.txt (line 1)) (from versions: none)
ERROR: No matching distribution found for argcomplete (from -r /home/rajesh/Developer/PX4-Autopilot/Tools/setup/requirements.txt (line 1))

```

MKexplorer, 2021-04-15:
> Hello,
> I followed the command which was mentioned [here](https://docs.px4.io/master/en/dev_setup/dev_env_linux_ubuntu.html) for Ubuntu 20.04 build but when I run `make px4_sitl jmavsim` or `make px4_sitl_defautl jmavsim`I get error.
> 
> Kindly check the last few lines of my terminal print with the error:
> 
> ```
> FAILED: generated_params/serial_params.c 
> cd /home/rajesh/Developer/PX4-Autopilot/build/px4_sitl_default/src/lib/parameters && /usr/bin/cmake -E make_directory /home/rajesh/Developer/PX4-Autopilot/build/px4_sitl_default/generated_params && /usr/bin/python3 /home/rajesh/Developer/PX4-Autopilot/Tools/serial/generate_config.py --params-file /home/rajesh/Developer/PX4-Autopilot/build/px4_sitl_default/generated_params/serial_params.c --serial-ports --ethernet --config-files /home/rajesh/Developer/PX4-Autopilot/src/lib/battery/module.yaml /home/rajesh/Developer/PX4-Autopilot/src/lib/pwm/pwm_main_params.yaml /home/rajesh/Developer/PX4-Autopilot/src/lib/pwm/pwm_aux_params.yaml /home/rajesh/Developer/PX4-Autopilot/src/lib/pwm/pwm_extra_params.yaml /home/rajesh/Developer/PX4-Autopilot/src/drivers/gps/module.yaml /home/rajesh/Developer/PX4-Autopilot/src/modules/mavlink/module.yaml
> Failed to import jinja2: No module named 'jinja2'
> 
> You may need to install it using:
>     pip3 install --user jinja2
> 
> ninja: build stopped: subcommand failed.
> make: *** [Makefile:221: px4_sitl_default] Error 1
> ```

I also got some errors when tried jmavsim for 1st time.

I just used `sudo make..`

julianoes, 2021-04-15:
> I just used sudo make..

Don't! That's the wrong approach.

Do what the error message tells you to do:

```
pip3 install --user jinja2
```

I just ran the ubuntu.sh script without error on Ubuntu 20.04.

@rajeshroy402 your issue looks like either pypi.org was down (or certificate expired), or something else in your networking not working correctly.

rajeshroy402, 2021-04-15:
> Don't! That's the wrong approach.
>
> Do what the error message tells you to do:
>
> ```
> pip3 install --user jinja2
> ```
>
> I just ran the ubuntu.sh script without error on Ubuntu 20.04.
>
> @rajeshroy402 your [...]

Hi, thank you for reaching out. May you explain what does (certificate expired) means?
```

ping pypi.org -c 10
PING pypi.org (151.101.128.223) 56(84) bytes of data.
64 bytes from 151.101.128.223 (151.101.128.223): icmp_seq=1 ttl=128 time=27.0 ms
64 bytes from 151.101.128.223 (151.101.128.223): icmp_seq=2 ttl=128 time=28.0 ms
64 bytes from 151.101.128.223 (151.101.128.223): icmp_seq=3 ttl=128 time=28.1 ms
64 bytes from 151.101.128.223 (151.101.128.223): icmp_seq=4 ttl=128 time=28.1 ms
64 bytes from 151.101.128.223 (151.101.128.223): icmp_seq=5 ttl=128 time=27.4 ms
64 bytes from 151.101.128.223 (151.101.128.223): icmp_seq=6 ttl=128 time=28.4 ms
64 bytes from 151.101.128.223 (151.101.128.223): icmp_seq=7 ttl=128 time=27.3 ms
64 bytes from 151.101.128.223 (151.101.128.223): icmp_seq=8 ttl=128 time=28.2 ms
64 bytes from 151.101.128.223 (151.101.128.223): icmp_seq=9 ttl=128 time=28.2 ms
64 bytes from 151.101.128.223 (151.101.128.223): icmp_seq=10 ttl=128 time=27.8 ms

--- pypi.org ping statistics ---
10 packets transmitted, 10 received, 0% packet loss, time 9020ms
rtt min/avg/max/mdev = 27.036/27.848/28.412/0.431 ms

```

Shouldn't be a problem with the network I guess...

Still, @julianoes , what do you suggest? How should I move forward now?

rajeshroy402, 2021-04-15:
Hi, I was able to launch the jMAVsim

Here's the summary before I close this issue.
After everything which didn't work from the above listed thread, I decided to remove the PX4-Autopilot local repository and clone it back again. 
I started getting error related to server certificates and then I followed my [this comment](https://github.com/ropensci/weathercan/issues/86#issuecomment-820267495) and reinstalled the certificates. 

After that, I was able to use the clone command and get a local repository. 
After cloning, I followed the regular steps mentioned on the developer site of px4 and this time I was able to get the setup up and running. 

Later, the `make px4_sitl_default jmavsim` command worked like a charm and the GUI launched.

Thank you!

closed 2021-04-15 by rajeshroy402