PX4/PX4-Autopilot #17398: SER_TEL1_BAUD parameter disappears when RTPS_CONFIG is selected

benjinne, 2021-04-14:
**Describe the bug**
When RTPS_CONFIG is selected to TELEM 1 instead of MAV_0_CONFIG, the parameter SER_TEL1_BAUD disappears. 

**To Reproduce**
Tested from latest master branch on modalai_fc-v1
Set MAV_0_CONFIG to disabled, and RTPS_CONFIG to TELEM 1, then SER_TEL1_BAUD disappears.

**Expected behavior**
Documentation states setting RTPS_CONFIG is a condition that enables the SER_TELX_BAUD parameter
https://docs.px4.io/master/en/peripherals/serial_configuration.html#how-to-configure-a-port

benjinne, 2021-04-14:
I found a solution! Using px4_get_parameter_value instead of stroul for the baudrate parameter in microRTPS_client_main.cpp activates the SER_TEL1_BAUD rate parameter somehow. I'm not sure how this works, but the reason I tried this was because it's how mavlink_main.cpp parses the parameters.

TSC21, 2021-04-14:
`RTPS_CONFIG` or `RTPS_MAV_CONFIG` is supposed to setup the module serial configuration that you find in https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/micrortps_bridge/micrortps_client/module.yaml.

benjinne, 2021-04-14:
> `RTPS_CONFIG` or `RTPS_MAV_CONFIG` is supposed to setup the module serial configuration that you find in https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/micrortps_bridge/micrortps_client/module.yaml.

Those parameters work as far as I know, however the baud rate isn't being set on startup since the SER_TEL1_BAUD parameter is getting marked as unused and set to 0. On startup, the debug console confirms this.

I think I've found a solution. Using px4_get_parameter_value() instead of stroul() for the baudrate parameter in microRTPS_client_main.cpp [here](https://github.com/PX4/PX4-Autopilot/blob/b99e597a739054f3814a4d8d26107b277ee8ac58/src/modules/micrortps_bridge/micrortps_client/microRTPS_client_main.cpp#L99) activates the SER_TEL1_BAUD rate parameter somehow. I'm not sure how this works, but the reason I tried this was because it's how mavlink_main.cpp parses the parameters [here](https://github.com/PX4/PX4-Autopilot/blob/b99e597a739054f3814a4d8d26107b277ee8ac58/src/modules/mavlink/mavlink_main.cpp#L1869).

TSC21, 2021-04-14:
> > `RTPS_CONFIG` or `RTPS_MAV_CONFIG` is supposed to setup the module serial configuration that you find in https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/micrortps_bridge/micrortps_client/module.yaml.
> 
> Those parameters work as far as I know, however the baud rate isn't being set on startup since the SER_TEL1_BAUD parameter is getting marked as unused and set to 0. On startup, the debug console confirms this.
> `INFO [micrortps_client] UART transport: device: /dev/ttyS6; baudrate: 0; sleep: 1ms; poll: 1ms; flow_control: No `
> 
> I think I've found a solution. Using px4_get_parameter_value() instead of stroul() for the baudrate parameter in microRTPS_client_main.cpp [here](https://github.com/PX4/PX4-Autopilot/blob/b99e597a739054f3814a4d8d26107b277ee8ac58/src/modules/micrortps_bridge/micrortps_client/microRTPS_client_main.cpp#L99) activates the SER_TEL1_BAUD rate parameter somehow. I'm not sure how this works, but the reason I tried this was because it's how mavlink_main.cpp parses the parameters [here](https://github.com/PX4/PX4-Autopilot/blob/b99e597a739054f3814a4d8d26107b277ee8ac58/src/modules/mavlink/mavlink_main.cpp#L1869).

Makes sense. But then that should be applied to the device as well. Can you issue a PR?

benjinne, 2021-04-14:
> Edit: maybe not for the device since it doesn't use the `p:` prefix. But I wonder if it is needed in this case or not. Do you mind verifying it?

By default the device is set to Disabled but it shows up in QGC and I can select it no problem

TSC21, 2021-04-14:
> > Edit: maybe not for the device since it doesn't use the `p:` prefix. But I wonder if it is needed in this case or not. Do you mind verifying it?
> 
> By default the device is set to Disabled but it shows up in QGC and I can select it no problem

Yeah can you try to set it as well and maybe even add the `p:` prefix in the yaml file. Just wanted to understand what exactly it happens. (It's not supposed to be set to disabled in my view)

benjinne, 2021-04-15:
> Yeah can you try to set it as well and maybe even add the `p:` prefix in the yaml file. Just wanted to understand what exactly it happens. (It's not supposed to be set to disabled in my view)

I think I understand how these parameters work now. The `p:` prefix isn't used for the `serial_device` argument since it's the same as the `port_config_param` in the `serial_config` section in the module.yaml. Mavlink sets `MAV_0_CONFIG` to `Telem1` [here](https://github.com/PX4/PX4-Autopilot/blob/2219e096d7bb8891976a421015dec4114317fa17/src/modules/mavlink/module.yaml#L35) and since there's no default parameter for `RTPS_CONFIG` [here](https://github.com/PX4/PX4-Autopilot/blob/2219e096d7bb8891976a421015dec4114317fa17/src/modules/micrortps_bridge/micrortps_client/module.yaml#L7) it is disabled by default.

benjinne, 2021-04-15:
I tried a quick easy fix by setting the parameters in the airframe:
```
MAV_0_CONFIG 0
RTPS_CONFIG 101
SER_TEL1_BAUD 460800
```

However, SER_TEL1_BAUD is still marked as unused. The only fix now is to `px4_get_parameter_value` like mavlink does [here](https://github.com/PX4/PX4-Autopilot/blob/b99e597a739054f3814a4d8d26107b277ee8ac58/src/modules/mavlink/mavlink_main.cpp#L1869) in micrortps_client_main.cpp. I'm working on a PR now

closed 2021-04-16 by benjinne