PX4/PX4-Autopilot #17410: HIL Support with fixed wing

LorenzMeier, 2021-04-16:
We should investigate what the current status is. There are user questions wether or not certain models work: https://docs.px4.io/master/en/simulation/hitl.html

vineethtom, 2021-04-16:
Thank you for your patience. I shall elaborate my problem.

I am trying to use firmware v1.9.0 on a Cuav V5+. I can see acuator controls in the Mavlink console in QGroundcontrol changing, but no servo movement. I am using a hitec d89mw servo in the m1 pin.

vineethtom, 2021-04-16:
The airframe i have been using is bormatec maja.

LorenzMeier, 2021-04-16:
Could you please follow the instructions and let me know where they fail?

https://docs.px4.io/master/en/simulation/hitl.html

Use the SITL Quad plane (config 1002) - to test the setup.

vineethtom, 2021-04-16:
Sorry i misguided you. This has nothing to do with HITL simulations. I
mentioned that just to give a heads up on why i have to stick to 1.9.0.

The problem statement is only about the servos not moving.

LorenzMeier, 2021-04-16:
1.9 is by now a very outdated release that we are not offering any support for. I shared the link to various HITL setups what are working with 1.11 and 1.12, please use those. I'm closing this issue in the meantime.

closed 2021-04-16 by LorenzMeier