PX4/PX4-Autopilot #17417: Pixracer outputs 8 times too hight pulse rates (current master)

taileron, 2021-04-17:
**Describe the bug**
Pulse rate 400Hz and 50Hz are present at 8 times the rate at all outputs.
Main rate is measured with osci 3.2KHz
default rate 400 Hz (instead of 50)
**To Reproduce**
Steps to reproduce the behavior:
as soon as one of my Pixracers are flashed 
default rate is displayed as 0 Hz
All servos immediately moved to the end stop, the ESCs give Error beep. 
**Expected behavior**
400Hz pwm rate gives 400Hz pulses
50Hz default rate gives 50Hz
**Log Files and Screenshots**
device: /dev/pwm_output0
channel 1: 900 us (alternative rate: 400 Hz failsafe: 0, disarmed: 900 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 2: 900 us (alternative rate: 400 Hz failsafe: 0, disarmed: 900 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 3: 900 us (alternative rate: 400 Hz failsafe: 0, disarmed: 900 us, min: 1118 us, max: 1943 us, trim:  0.00)
channel 4: 2000 us (alternative rate: 400 Hz failsafe: 0, disarmed: 900 us, min: 976 us, max: 2011 us, trim:  0.00)
channel 5: 1985 us (default rate: 0 Hz failsafe: 1985, disarmed: 1985 us, min: 650 us, max: 2450 us, trim:  0.00)
channel 6: 1030 us (default rate: 0 Hz failsafe: 1030, disarmed: 1030 us, min: 650 us, max: 2450 us, trim:  0.00)
channel group 0: channels 1 2 3 4
channel group 1: channels 5 6

logging creates only corrupt data

dagar, 2021-04-17:
- which airframe did you select (SYS_AUTOSTART)?
 - `param show PWM*`
 - `pwm_out status`

taileron, 2021-04-17:
doesn´t matter it´s on all Fcs and Airframes there ist something with frequencies in the FMU.
I tried vtols like Convergence on Pixracer or X21-777  or Tailsitters 
 on Omnibus. 
If the IO is switched of, the FMU on X2.1-777 gives the same.

dagar, 2021-04-17:
Thanks for the detail, fixing this is my top priority.

taileron, 2021-04-17:
@dagar The same pulse pattern also occurs when the master of 31 March is switched from 400Hz to Oneshot for a short time and then back to 400Hz again.
pwm rate -g 1 -r 400
pwm rate -g 1 -r 0 
pwm rate -g 1 -r 400
corresponds to the OneShot timer but with 400Hz PWM setting results in 3.2 - 3.35 Khz

taileron, 2021-04-18:
To eliminate such errors completely, one could throw FMU OneShot overboard, as FMU dshot works extremely well. (At the IO it still makes sense, since dshot is not there) OneShot should output a constant frequency up to approx. 4 KHz. With all previous implementations, the output at the FMU tries to orientate itself to imu_gyro_rm, but does not never succeed and outputs a very unstable frequency signal between 400 and 700Hz, with which many ESC are not satisfied so it offers no advantages over 400Hz standard PWM.

LorenzMeier, 2021-04-18:
Thanks for reporting and the instructions to reproduce!

dagar, 2021-04-19:
I'm not sure about the OneShot issues, but I think I understand what's happening with the secondary PWM rate configuration. Let me see about a short term fix.

We're working towards being able to explicitly configure all outputs directly (including conveying rate groups constraints).

taileron, 2021-04-19:
The problem with OneShot then and now is mainly when servos with 50Hz are needed on another output group at the same time.
With Airframe Generic Quad only ESC Out, DSHOT and OneShot with 1KHz works with the current error without any problems including successful flight tests. Only standard PWM e.g. 400Hz gives OneShot with 3.2KHz clock frequency which only controls the clockwise rotating motors successfully.

taileron, 2021-04-23:
#17427 solves the PWM issue for me, any combination of default and alternative rates now offer the right output pulses. 
The incorrect 3.2Khz behaviour is only still present if you briefly switch to OneShot and then back to alternative rate.
pwm rate -g 1 -r 400
pwm rate -g 1 -r 0
pwm rate -g 1 -r 400

AlexKlimaj, 2021-04-24:
I can confirm this bug on 1.12 beta 3. Set to 100Hz and measures 800Hz on the scope.

MaEtUgR, 2021-04-28:
Related: https://github.com/PX4/PX4-Autopilot/pull/17427

closed 2021-06-25 by CUAVcaijie