PX4/PX4-Autopilot #17419: RC calibrated : Detected 0 radio channels. To operate PX4, you need at least 5 channels.

yinjiajie, 2021-04-19:
I use PX4 DurandalV1, the RC can be calibrated if selecting Generic Quadcopter airframe, but RC can not be calibrated if selecting Blade 130X airframe, display "Detected 0 radio channels. To operate PX4, you need at least 5 channels".

bresch, 2021-04-19:
Do you have a RC receiver connected? To SBUS? Spektrum? How did connect it?

yinjiajie, 2021-04-19:
yes, i have a RC receiver connected to SBUS. If selecting Generic Quadcopter airframe, it is OK. Then, i just modify airframe to Blade 130X, it is not OK.

yinjiajie, 2021-04-19:
[image]

taileron, 2021-04-23:
Some mixers that use a little more memory, especially when PWM output limits have been extended above 2000uS, have sometimes blocked RC detection.
At the moment, I only have this under control by loading the mixer later in extras.txt.

taileron, 2021-05-02:
with the current master changing src/modules/px4iofirmware/mixer.h
mixer lenght from 330 to 430 now allows to load all mixers successfully
without blocking serial rc anymore

dagar, 2021-05-02:
@taileron can you check free memory (px4io status) while using a mixer and RC? Let's make that change right away if it fits.

taileron, 2021-05-03:
how much memory should be left free in the IO?
Status is from PixFalcon where so far serial RC was always blocked when I wanted to load my Vectored Thrust Mixer with 5 arguments and mixer length = 530. With ml=430 the mixer didn't´t load or loaded not completely. 
At the moment it works all without problems.

 px4io_status_s
	timestamp: 106446396  (0.508539 seconds ago)
	voltage_v: 0.0000
	rssi_v: 1.1850
	free_memory_bytes: 712
	actuators: [0, 0, 0, 9808, 4841, -5769, 0, 0]
	servos: [1000, 1000, 1000, 2000, 1985, 1030, 1500, 1500]
	raw_inputs: [1500, 1501, 998, 1500, 1697, 1220, 1945, 2096, 1057, 1898, 1762, 1500, 1761, 1500, 0, 0, 0, 0]
	status_outputs_armed: True
	status_override: False
	status_rc_ok: True
	status_rc_ppm: False
	status_rc_dsm: True
	status_rc_sbus: False
	status_fmu_ok: True
	status_raw_pwm: False
	status_mixer_ok: True
	status_arm_sync: True
	status_init_ok: True
	status_failsafe: False
	status_safety_off: True
	status_fmu_initialized: True
	status_rc_st24: False
	status_rc_sumd: False
	alarm_vbatt_low: False
	alarm_temperature: False
	alarm_servo_current: False
	alarm_acc_current: False
	alarm_fmu_lost: True
	alarm_rc_lost: True
	alarm_pwm_error: False
	alarm_vservo_fault: False
	arming_io_arm_ok: True
	arming_fmu_armed: False
	arming_fmu_prearmed: True
	arming_manual_override_ok: False
	arming_failsafe_custom: True
	arming_inair_restart_ok: True
	arming_always_pwm_enable: True
	arming_rc_handling_disabled: False
	arming_lockdown: False
	arming_force_failsafe: False
	arming_termination_failsafe: False
	arming_override_immediate: True

reversed outputs: [________] trims: r:   0.0000 p:   0.0000 y:   0.0000
14 raw R/C inputs 1500 1501 998 1500 1697 1220 1945 2096 1057 1898 1762 1500 1761 1500
R/C flags: 0x001c DSM11 MAPPING_OK
mapped R/C inputs 0x00af 0:0 1:0 2:0 3:0 5:-10000 7:0
ADC inputs 0 1743
features 0x0008 RSSI_ADC
rates 0x000f default 50 alt 400 sbus 72
debuglevel 0
controls 0: -224 59 2 0 0 0 0 0
controls 1: 0 0 0 0 0 0 0 0
controls 2: 0 0 0 0 0 0 0 0
controls 3: 0 0 0 0 0 -10000 0 0
input 0 min 915 center 1500 max 2082 deadzone 10 assigned 0 options 0x0003 ENABLED REVERSED
input 1 min 900 center 1500 max 2096 deadzone 10 assigned 1 options 0x0001 ENABLED
input 2 min 1031 center 1031 max 2101 deadzone 10 assigned 3 options 0x0001 ENABLED
input 3 min 908 center 1500 max 2094 deadzone 10 assigned 2 options 0x0003 ENABLED REVERSED
input 4 min 1000 center 1525 max 2000 deadzone 10 assigned 255 options 0x0000
input 5 min 1039 center 1500 max 1961 deadzone 10 assigned 255 options 0x0000
input 6 min 899 center 1500 max 1921 deadzone 10 assigned 5 options 0x0003 ENABLED REVERSED
input 7 min 800 center 1500 max 2200 deadzone 10 assigned 255 options 0x0000
input 8 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 9 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0002 REVERSED
input 10 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0002 REVERSED
input 11 min 1099 center 1500 max 1901 deadzone 0 assigned 7 options 0x0001 ENABLED
input 12 min 1099 center 1500 max 1901 deadzone 0 assigned 255 options 0x0000
input 13 min 1099 center 1500 max 1901 deadzone 0 assigned 255 options 0x0000
input 14 min 1099 center 1500 max 1901 deadzone 0 assigned 255 options 0x0000
input 15 min 1099 center 1500 max 1901 deadzone 0 assigned 255 options 0x0000
input 16 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 17 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
failsafe 0 0 0 2000 1985 1030 1500 1500
disarmed values 1000 1000 1000 2000 1985 1030 1500 1500
nsh>

Abner0907, 2021-06-11:
The FC have not got the rc receiver's value. So how can you calibrate the rc

XinLong-Li, 2023-04-10:
I think you may need to check some parament in QGroundControl, the first one is `RC_INPUT_PROTO`
[image]

closed 2025-12-03 by github-actions[bot]