PX4/PX4-Autopilot #17420: Gumstix FMU-V6U battery status not recognized in QGC

473867143, 2021-04-19:
I'm from the AMOVlab (a Dronecode member) R&D team, and we are testing the Gumstix FMUv6U board. We have put the board on our P230 frame, but QGC cannot recognize battery status. To take-off, we have to manually turn-off the CBRK_SUPPLY_CHK power supply circuit breaker to unlock/arm the vehicle. 

[image]

This screenshot shows for the power supply step, there is no battery detected.
[image]

The following screenshots showing the failure of calibration:

[image]

[image]

[image]

We are suspecting there is power supply pinout inconsistency, but not sure. In the [Gumstix documentation ](https://github.com/gumstix/PKG900000001531/blob/master/PKG900000001531_AutoDoc.pdf) 1.8.2 it shows the 6-pin JST but module not connected. 

[image]

Is it the same as the FMUv5 pin-out as below?
[image]

473867143, 2021-04-19:
@smithandrewc  do you have the same issue?

473867143, 2021-04-19:
In the hardware schematic on [Gumstix's doc Page 24](https://github.com/gumstix/PKG900000001531/blob/master/hardware/PKG900000001531.pdf), it shows the JST 6 pin;s pin3 is 6V. Is that correct by design? 
[image]

smithandrewc, 2021-04-19:
Hi,

Sorry you're having issues.  That schematic is correct.  Pin 3 isn't 6V, that's just the pin name given in the pin-out specification.  We elected not to setup a voltage divider to enable max 6.6V input since we're using it for current sensing.  What power module are you using?

jinger26, 2021-04-19:
@smithandrewc  is your battery status functional in QGC?

jinger26, 2021-04-19:
@473867143  is using a power module that typically runs v5 (pixhawk 4/4mini)

smithandrewc, 2021-04-19:
I did test it initially but I need to reconfirm.  @473867143, did you reflash your FMUv6U after receiving it or is it still running the image it was shipped with?

473867143, 2021-04-20:
@smithandrewc  I didn‘t flash any firmware

smithandrewc, 2021-04-20:
Ok thanks, I'll confirm functionality as soon as possible.

smithandrewc, 2021-04-22:
Small update.  I confirmed the tests I did when we released the board.  

https://github.com/PX4/PX4-Autopilot/blob/master/boards/px4/fmu-v6u/src/board_config.h#L152

ADC3_3V3 (ADC 18) is setup to measure the battery voltage and ADC3_6V6 (ADC 12) is setup to measure the current.  If you log into the mavlink console (plug in a USB-C and open a terminal @ 57600 to the first new serial device) you can run the command "board_adc test".  That will show the currently configured ADC channels.  In the list you will see channel 12 and channel 18.  While the test is running you can change the voltage (pull it to GND) and see that reflected in the results.  So the pins are connected appropriately.

However, two issues I'm running into

- I'm not seeing those ADC values passed into QGroundcontrol.
- I've plugged in a PM02 v4.1 into a 3S battery and I'm seeing 3.3V on both the current and voltage pins (3, 4) coming out of the PM02.  I would have expected this to about more like 0.632493675 V (assuming 18.182 voltage divider).

I'm working on it.  If you have any thoughts or ideas, I would love to hear them.  Thanks.

smithandrewc, 2021-04-22:
I have another quick update.  You need to update a couple of parameters.  I will make the change in the firmware so in future releases these are automatic but for now you can do it manually.  Change the following:

- BAT1_I_CHANNEL = 12
- BAT1_V_CHANNEL = 18

See below on how you'd do it in QGroundControl

[image]

Once you've made the change and rebooted the vehicle you can confirm the change by logging into the console and running the following command

`nsh> listener battery_status`

That should show something like this

`nsh> listener battery_status

TOPIC: battery_status
 battery_status_s
        timestamp: 521066218  (0.006083 seconds ago)
        voltage_v: 33.7841
        voltage_filtered_v: 33.3192
        current_a: 27.4779
        current_filtered_a: 26.0915
        average_current_a: 0.0000
        discharged_mah: 3933.3374
        remaining: 1.0000
        scale: 1.0000
        temperature: nan
        cell_count: 3
        voltage_cell_v: [11.1064, 11.1064, 11.1064, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000]
        max_cell_voltage_delta: 0.0000
        capacity: 0
        cycle_count: 0
        run_time_to_empty: 0
        average_time_to_empty: 0
        serial_number: 0
        manufacture_date: 0
        state_of_health: 0
        max_error: 0
        interface_error: 0
        connected: False
        source: 0
        priority: 0
        id: 1
        is_powering_off: False
        warning: 0
`
My numbers are way off because the battery isn't connected but they're not reading 0.  I'm still looking into why the PM02 is putting out that voltage.  I'll report back when I have something.

473867143, 2021-04-23:
@smithandrewc  THs for your reply!

I will reproduce it as soon as possible.

@jinger26  hello, Shall we test it again next week?

mrpollo, 2021-04-28:
@473867143 which version of PX4 are you using?

closed 2025-12-03 by github-actions[bot]