PX4/PX4-Autopilot #17423: MC idle PWM automatically changes to 1000ms

VTOLDavid, 2021-04-19:
It normally arms in MC mode properly and sets pwm to 1100 while idle. (Where is the parameter to adjust this?) But sometimes arms and pwm motor signals stay at 1000ms, thus motors remain stopped. If throttle is increased It responds but ESCs turn off if the signal returns to 1000ms and you get erratic behaviour from motors.

To Reproduce, not easily reproducible...
-Arm MC everithing OK
-Disarm
-Let the drone disarmed for a while (20min aprox)
-Arm, it arms but motors do not spin at iddle.

Expected behavior
This should not happen.

Log Files and Screenshots
[FlightLog](https://logs.px4.io/plot_app?log=eef8b1e5-baa1-4bd3-84fc-0ba256c10845)

Drone:
Firmware: 1.11.0 Master
Airframe: Tailsitter
Autopilot: Cube black

bresch, 2021-04-23:
Are you aware of this @sfuhrer ?

sfuhrer, 2021-04-26:
> Are you aware of this @sfuhrer ?

No, it's been years since I last flew a tailsitter.

sfuhrer, 2021-04-26:
@VTOLDavid did you per chance also try to reproduce it in SITL?

VTOLDavid, 2021-04-27:
I tried but is not easy to reproduce.