PX4/PX4-Autopilot #17442: SD Logging aborts on all boards shortly after start-up ERROR [logger] write failed (116)

taileron, 2021-04-22:
**Describe the bug**
After arming the SD logging stops after a short period of time

**To Reproduce**
Steps to reproduce the behavior:
happens on all tested Fc 

**Expected behavior**
A clear and concise description of what you expected to happen.
A proper logfile gets created according to the sd log settings 
**Log Files and Screenshots**
*Always* provide a link to the flight log file:
HW arch: PX4_FMU_V4
FW git-hash: 38bc46fd1af5823c58a84abc7c08172e82a85039
FW version: 1.12.0 0 (17563648)
FW git-branch: master
OS: NuttX
OS version: Release 10.0.0 (167772415)
OS git-hash: 405cfa2571ab9eee2fd8cdf28726f8ccfeb3e5e6
Build datetime: Apr 17 2021 16:15:02
Build uri: localhost
Toolchain: GNU GCC, 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
PX4GUID: 00010000000036313737333951050051002d
MCU: STM32F42x, rev. 5
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [parameters] BSON document size 3856 bytes, decoded 3856 bytes
Board architecture defaults: /etc/init.d/rc.board_arch_defaults
Board defaults: /etc/init.d/rc.board_defaults
INFO  [dataman] Unknown restart, data manager file '/fs/microsd/dataman' size is 362560 bytes
Board sensors: /etc/init.d/rc.board_sensors
ms5611 #0 on SPI bus 2 (devid=0x3d)
WARN  [SPI_I2C] hmc5883: no instance started (no device on bus?)
lis3mdl #0 on SPI bus 1 (devid=0x5)
icm20602 #0 on SPI bus 1 (devid=0x38)
mpu9250 #0 on SPI bus 1 (devid=0x24)
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
Starting Main GPS on /dev/ttyS3
INFO  [pwm_out] instance: 0, max rate: 100, default: 50, alt: 50
INFO  [init] Mixer: /etc/mixers/vtol_convergence.main.mix on /dev/pwm_output0
ekf2 [495:100]
INFO  [pwm_out] instance: 0, max rate: 800, default: 50, alt: 400
Board extras: /etc/init.d/rc.board_extras
Addons script: /fs/microsd/etc/extras.txt
INFO  [logger] logger started (mode=all)

NuttShell (NSH) NuttX-10.0.0
nsh> INFO  [mavlink] Starting mavlink shell
INFO  [commander] Armed by external command
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] /fs/microsd/log/2021-04-22/11_18_20.ulg
INFO  [logger] Opened full log file: /fs/microsd/log/2021-04-22/11_18_20.ulg
ERROR [logger] write failed (116)
WARN  [logger] closing log file failed (116)
INFO  [commander] Disarmed by external command
INFO  [commander] Connection to ground station lost
INFO  [commander] Data link regained

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

taileron, 2021-04-22:
I found a small hint, all my projects on various flight controllers always need an extended PWM limit upwards, PWM Max 2350uS or 2450uS. 
In order to use the firmware at all, I have to change PWM_HIGHEST_MAX in drv_pwm_output.h from 2150 to 2500 first. However, the 1.12 logger no longer seems to be able to process this, especially if the corresponding values also took place.

bkueng, 2021-04-23:
```
ERROR [logger] write failed (116)
```
@davids5 I'm seeing this timeout as well on a Pixracer with cheap SD, randomly after logging starts. In one case the SD got completely inaccessible afterwards:
```
nsh> cd /fs/microsd
nsh> ls
/fs/microsd:
nsh>
```

Reverting #17350 does not help.

taileron, 2021-04-23:
This problem was there before #17350, it has existed since mid-January but I thought it was a Ram overflow due to UAVCAN tests until I noticed it appears identically on all FCs I have even without UAVCAN and not all of them have inferior SD cards. Omnibus F4 (F404), Pixracer/Pixfalcon (F427), X2.1-777 (F777). If corrupt data was written to it making it inaccessible, after a reboot of the controller, it appears again without recognisable errors after attempt of data medium repair on the PC.

taileron, 2021-04-26:
After I removed all the lines for "prevent Mac and Ubuntu from creating un .... index files" in rcS and reformatted the SD to Fat32 on the PC (not Mac !!!!), logging is possible again in all my FCs if it is a current 8...16Gb SD. PWM limits etc. do not play a major role.
Unfortunately, with the current master, all the 1 Gb SDs that came with the PixRacer R12, R14 and PixFalcon can no longer be used for logging, even after reformatting to Fat32, although FTP access and the use of alternative mixers, startups etc. works without any problems.

EliaTarasov, 2021-04-27:
SD: SanDisk Extreme U3 32GB, FAT32 formatted (tried both Ubuntu and Windows).
Pixhawk: 2.4.8.
PX4: v1.11.3, v.1.12.0 beta1-2, master.

nsh> logger stop
nsh> logger start
INFO  [logger] logger started (mode=all)
nsh> logger status
INFO  [logger] Running in mode: all
INFO  [logger] Number of subscriptions: 162 (5184 bytes)
INFO  [logger] Not logging
nsh> cd /fs/microsd/log
nsh> ls
/fs/microsd/log:
nsh>

taileron, 2021-04-27:
@EliaTarasov That looks good, as soon as armed it starts to log (or SDLOG_MODE = 1 or 2):
nsh> logger status
INFO  [logger] Running in mode: all
INFO  [logger] Number of subscriptions: 170 (5440 bytes)
INFO  [logger] Full File Logging Running:
INFO  [logger] Log file: /fs/microsd/log/2021-04-27/18_01_47.ulg
INFO  [logger] Wrote 0.47 MiB (avg 40.32 KiB/s)
INFO  [logger] Since last status: dropouts: 4 (max len: 0.021 s), max used buffer: 7148 / 8192 B
nsh> cd /fs/microsd/log/2021-04-27
nsh> ls
/fs/microsd/log/2021-04-27:
18_01_47.ulg
nsh>

EliaTarasov, 2021-04-27:
Great!
What is exactly your SW/HW configuration, please?

taileron, 2021-04-27:
it´s master from 5 days ago tested PixRacer or MRO X21-777 SD cards are SanDisk 8Gb
In the versions between 11.3 and 12 Beta, logging did not work for me from time to time.
But I just noticed that as soon as UAVCAN is activated, the log buffer scheme changes and logging again no longer works ...

EliaTarasov, 2021-04-28:
> it´s master from 5 days ago tested PixRacer or MRO X21-777 SD cards are SanDisk 8Gb

I see. I've just tested 1.11.0 and the problem remains. 
So in addition to the initial issue it might also be 32GB problem.
Do you have 32 GB SD to test?

taileron, 2021-04-28:
32 Gb not yet tested, but as soon as the problem occurred once, the file system of the SD is corrupt and has to be reformatted, otherwise the logger will not work anymore at all, even if everything would be OK again from the firmware site. My 8 and 16Gb SD have to be fat32 formatted in 4096 blocks size to work correctly. Some of the 1Gb only work with 512 block size.

taileron, 2021-04-28:
X2.1-777 seems to have fewer problems now because the buffer scheme is very generous. With Pixfalcon, Omnibus F4 and PixRacer with or without UAVCAN, no buffering runs after 3 Mb at the latest and usually the file system is no longer accessible. Better SD cards produce fewer dropouts, but do not prevent the problem.
On PixRacer, the log buffer size 9 k is best.
The automatic reduction to 4 k, if UAVCAN is activated, prevents all logging completely, but the size 9 k still leaves enough ram. Then I have different PixRacers (R12,R14,R15) that react differently, but with all of them the file system is inaccessible at some point after logging.

EliaTarasov, 2021-04-28:
> 32 Gb not yet tested, but as soon as the problem occurred once, the file system of the SD is corrupt and has to be reformatted, otherwise the logger will not work anymore at all, even if everything would be OK again from the firmware site.

Looks like 32 GB is wrong choice, but it is [recommended](https://docs.px4.io/master/en/dev_log/logging.html). Will try to format with different blocks size. Thanks for clarification.

taileron, 2021-04-28:
SD_BENCH can be used to test the SD interface itself. This is where the actual cause lies - after a certain number of write operations, access to the SD is completely lost. Different SD cards have different values and different durations until the sd file system is gone. 
 
nsh> sd_bench
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:   709.20 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 21 ms
INFO  [sd_bench]   Run  1:   689.01 KB/s, max write time: 58 ms (=  68.97 KB/s), fsync: 19 ms
ERROR [sd_bench] Write error

taileron, 2021-04-29:
from the other PixRacer:
nsh> free
             total       used       free    largest
Umem:       226880     181504      45376      45008
Prog:      2097152    1835008     262144          0
nsh> sd_bench
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:   447.21 KB/s, max write time: 68 ms (=  58.82 KB/s), fsync: 101 ms
INFO  [sd_bench]   Run  1:   442.57 KB/s, max write time: 87 ms (=  45.98 KB/s), fsync: 76 ms
INFO  [sd_bench]   Run  2:   443.53 KB/s, max write time: 88 ms (=  45.45 KB/s), fsync: 77 ms
INFO  [sd_bench]   Run  3:   441.47 KB/s, max write time: 86 ms (=  46.51 KB/s), fsync: 26 ms
INFO  [sd_bench]   Run  4:   432.67 KB/s, max write time: 86 ms (=  46.51 KB/s), fsync: 24 ms
INFO  [sd_bench]   Avg   :   441.55 KB/s
nsh>  -r 50
nsh: -r: command not found
nsh> sd_bench -r 50
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:   463.55 KB/s, max write time: 88 ms (=  45.45 KB/s), fsync: 24 ms
INFO  [sd_bench]   Run  1:   462.75 KB/s, max write time: 113 ms (=  35.40 KB/s), fsync: 25 ms
INFO  [sd_bench]   Run  2:   480.89 KB/s, max write time: 22 ms (= 181.82 KB/s), fsync: 77 ms
INFO  [sd_bench]   Run  3:   462.65 KB/s, max write time: 34 ms (= 117.65 KB/s), fsync: 74 ms
ERROR [sd_bench] Write error
nsh> cd /fs/microsd
nsh> ls
/fs/microsd:

LorenzMeier, 2021-04-29:
@davids5 This needs attention and looks like a release blocker. @mrpollo FYI

mrpollo, 2021-04-29:
We'll look into this ASAP!

taileron, 2021-04-30:
now I got a SanDisk Extreme 32Gb to test, it happens less frequently:
And there I also have a little hint, sd freshly unpacked sd_bench ran through over 50 times without errors but as soon as the SD was written and data has to be overwritten, the problems could come because flash media have to be erased prior  written to and erase block size differs from write blocks.
Most problems occur with SanDisk EDGE,  or ATP which were fully written once.
nsh> free (lower free ram cause UAVCAN enabled)
             total       used       free    largest
Umem:       226240     205424      20816      20576
Prog:      2097152    1835008     262144          0
nsh> sd_bench -r 55
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:  1699.92 KB/s, max write time: 45 ms (=  88.89 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run  1:  1758.36 KB/s, max write time: 8 ms (= 500.00 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run  2:  1717.39 KB/s, max write time: 16 ms (= 250.00 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run  3:  1697.61 KB/s, max write time: 47 ms (=  85.11 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run  4:  1722.44 KB/s, max write time: 19 ms (= 210.53 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run  5:  1693.99 KB/s, max write time: 44 ms (=  90.91 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run  6:  1684.11 KB/s, max write time: 46 ms (=  86.96 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run  7:  1728.47 KB/s, max write time: 56 ms (=  71.43 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run  8:  1717.63 KB/s, max write time: 44 ms (=  90.91 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run  9:  1744.09 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 10:  1720.13 KB/s, max write time: 44 ms (=  90.91 KB/s), fsync: 10 ms
INFO  [sd_bench]   Run 11:  1713.46 KB/s, max write time: 15 ms (= 266.67 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 12:  1706.27 KB/s, max write time: 44 ms (=  90.91 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 13:  1684.86 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 14:  1676.80 KB/s, max write time: 53 ms (=  75.47 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 15:  1730.70 KB/s, max write time: 16 ms (= 250.00 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 16:  1737.42 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 17:  1671.36 KB/s, max write time: 46 ms (=  86.96 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 18:  1746.61 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 19:  1722.98 KB/s, max write time: 46 ms (=  86.96 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run 20:  1723.80 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 21:  1666.80 KB/s, max write time: 59 ms (=  67.80 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 22:  1777.18 KB/s, max write time: 8 ms (= 500.00 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 23:  1697.44 KB/s, max write time: 44 ms (=  90.91 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 24:  1729.09 KB/s, max write time: 8 ms (= 500.00 KB/s), fsync: 6 ms
ERROR [sd_bench] Write error
nsh> cd fs/microsd
nsh> ls
/fs/microsd:
nsh>

davids5, 2021-04-30:
```

taileron, 2021-04-30:
@davids5 It only occurs on STM F427 MCU (e.g. PixRacer/PixFalcon) I have tested it on F7xx (e.g. X2.1-777) and F405 (e.g. Omnibus F4) there is no Problem at all even with extremely slow, filled or old sd cards. 
nsh> ls
/fs/microsd:
 System Volume Information/
 etc/
 dataman
 ufw/
 uavcan.db/
 log/
nsh> dmesg
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: PX4_FMU_V4
FW git-hash: d631a5d39f4bdf0acb1ef5973616633e179877d3
FW version: 1.12.0 0 (17563648)
FW git-branch: master
OS: NuttX
OS version: Release 10.0.0 (167772415)
OS git-hash: 405cfa2571ab9eee2fd8cdf28726f8ccfeb3e5e6
Build datetime: Apr 28 2021 11:45:18
Build uri: localhost
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 00010000000036313737333951050051002d
MCU: STM32F42x, rev. 5
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [parameters] BSON document size 3914 bytes, decoded 3914 bytes
Board architecture defaults: /etc/init.d/rc.board_arch_defaults
Board defaults: /etc/init.d/rc.board_defaults
INFO  [dataman] Unknown restart, data manager file '/fs/microsd/dataman' size is 362560 bytes
INFO  [uavcan] Node ID 1, bitrate 1000000
Board sensors: /etc/init.d/rc.board_sensors
ms5611 #0 on SPI bus 2 (devid=0x3d)
WARN  [SPI_I2C] hmc5883: no instance started (no device on bus?)
lis3mdl #0 on SPI bus 1 (devid=0x5)
icm20602 #0 on SPI bus 1 (devid=0x38)
mpu9250 #0 on SPI bus 1 (devid=0x24)
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
Starting Main GPS on /dev/ttyS3
INFO  [pwm_out] instance: 0, max rate: 100, default: 50, alt: 50
INFO  [init] Mixer: /fs/microsd/etc/mixers/vtol_convergence.main.mix on /dev/pwm_output0
ekf2 [527:100]
INFO  [pwm_out] instance: 0, max rate: 800, default: 50, alt: 400
Board extras: /etc/init.d/rc.board_extras
Addons script: /fs/microsd/etc/extras.txt
INFO  [logger] logger started (mode=all)

EliaTarasov, 2021-04-30:
@davids5 @taileron looks like 16kb cluster size did the trick. It finally started to write complete log on master. At least for the Pixhawk 2.4.8. 
[image]
But the average speed I guess may be increased. 
[image]

taileron, 2021-04-30:
@EliaTarasov here´s the default cluster size of windows managed volumes:
https://support.microsoft.com/en-us/topic/default-cluster-size-for-ntfs-fat-and-exfat-9772e6f1-e31a-00d7-e18f-73169155af95
so 32Gb should theorethical offer 16kb blocks before the first formatting. My SD Extreme offered 32K block size at the first use, displayed by chkdsk (third last line) and I didn´t touch any format procedure

davids5, 2021-04-30:
@EliaTarasov - that is good to hear.

I just ran the same test as above with the Extreme card on the FMUV4 (Pixracer RC15)  it is ALL good.
```
nsh> [boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: PX4_FMU_V4
FW git-hash: 4f52c0b6daa93f01b3c02e9457265be107af74be
FW version: 1.12.0 80 (17563776)
FW git-branch: master
OS: NuttX
OS version: Release 10.0.[hardfault_log] Fault Log is Armed
0 (167772415)
OS git-hash: 405cfa2571ab9eee2fd8cdf28726f8ccfeb3e5e6
Build datetime: Apr 30 2021 08:09:54
Build uri: localhost
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 0001000000003535323035375110003e0039
MCU: STM32F42x, rev. 3
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [parameters] BSON document size 305 bytes, decoded 305 bytes
Board architecture defaults: /etc/init.d/rc.board_arch_defaults
Board defaults: /etc/init.d/rc.board_defaults
INFO  [dataman] Unknown restart, data manager file '/fs/microsd/dataman' size is 362560 bytes
Board sensors: /etc/init.d/rc.board_sensors
reset done, 50 ms
ms5611 #0 on SPI bus 2 (devid=0x3d)
WARN  [SPI_I2C] hmc5883: no instance started (no device on bus?)
lis3mdl #0 on SPI bus 1 (devid=0x5)
WARN  [SPI_I2C] icm20602: no instance started (no device on bus?)
icm20608g #0 on SPI bus 1 (devid=0x3a)
mpu9250 #0 on SPI bus 1 (devid=0x24)
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
Starting Main GPS on /dev/ttyS3
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
Starting MAVLink on /dev/ttyS0
INFO  [mavlink] mode: Normal, data rate: 20000 B/s on /dev/ttyS0 @ 921600B
INFO  [pwm_out] instance: 0, max rate: 100, default: 50, alt: 50
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
INFO  [init] setting PWM_AUX_OUT none
ekf2 [445:100]
INFO  [pwm_out] instance: 0, max rate: 800, default: 50, alt: 400
Board extras: /etc/init.d/rc.board_extras
INFO  [logger] logger started (mode=all)
```

Reformatted it to 32K AUs and it still passed

Put 2.6 Gb of files on it and it is still passing

```
INFO  [logger] [logger] /fs/microsd/log/sess003/log001.ulg
INFO  [logger] Opened full log file: /fs/microsd/log/sess003/log001.ulg

nsh>
nsh>
nsh> sd_bench
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench]
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:  1810.23 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run  1:  1783.61 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 4 ms
INFO  [frsky_telemetry] Scanning timeout: exiting
INFO  [sd_bench]   Run  2:  1743.68 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run  3:  1775.64 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run  4:  1776.13 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 5 ms
INFO  [sd_bench]   Avg   :  1777.86 KB/s
```

Questions for you @taileron @EliaTarasov both:

1) Does this happen to many of the exact same cards or is it just one?
2) Is it repeatable with changing the format to and from working?

EliaTarasov, 2021-04-30:
1. I tested only one card. In a next couple of days (I guess) there will be more information.
2. Yes, I formatted and tested several times for 64KB, 32KB, 16KB, 8KB and 4KB. 
    32, 16 and 8 seem stable on all iterations.

taileron, 2021-05-01:
With the SanDisk Extreme 32Gb it happens rarely enough to be able to live with it. sd_bench usually runs through over 75 times (PixRacer R12). Ordered some more now.
HW arch: PX4_FMU_V4
FW git-hash: d631a5d39f4bdf0acb1ef5973616633e179877d3
FW version: 1.12.0 0 (17563648)
FW git-branch: master
OS: NuttX
OS version: Release 10.0.0 (167772415)
OS git-hash: 405cfa2571ab9eee2fd8cdf28726f8ccfeb3e5e6
Build datetime: Apr 30 2021 23:11:45
Build uri: localhost
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000100000000343739393137511400540034
MCU: STM32F42x, rev. 3
nsh> sd_bench -r 99
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:  1282.76 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run  1:  1244.81 KB/s, max write time: 19 ms (= 210.53 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run  2:  1270.15 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run  3:  1270.72 KB/s, max write time: 15 ms (= 266.67 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run  4:  1232.22 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run  5:  1260.58 KB/s, max write time: 20 ms (= 200.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  6:  1249.72 KB/s, max write time: 19 ms (= 210.53 KB/s), fsync: 8 ms

INFO  [sd_bench]   Run 97:  1221.01 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 98:  1225.22 KB/s, max write time: 16 ms (= 250.00 KB/s), fsync: 20 ms
INFO  [sd_bench]   Avg   :  1253.27 KB/s

All SanDisk Edge 8Gb, ATP and Apacer 16Gb lose the file system in the log process after less than 60 sec in PicRacer and PixFalcon especially when UAVCAN is enabled. SanDisk Ultra last a little longer. 
But these SDs run without problems in the F777 and F405 Fc that´s why I initially opened the issue.

EliaTarasov, 2021-05-04:
T-Flash MicroSD 2GB writes ~30 min high resolution log with 16KB and 8 KB cluster size.
Stable repeatability.

[image]

davids5, 2021-05-04:
@EliaTarasov @taileron 

At this point what is not working?

EliaTarasov, 2021-05-05:
@davids5 it seems to work okay now, with some additional notes:
1. Given T-Flash seems that cluster size only affects write speed (slightly).
2. On the contrary SanDisk Extreme 32 GB is susceptible to cluster size choice.

EliaTarasov, 2021-05-05:
@taileron  did you try your SanDisk 32 GB to write log with sufficient enough duration (~30 minutes)?

taileron, 2021-05-05:
For further testing I am waiting for the next SanDisk Extreme as I had to glue the one I had back in the plane where the SD slot was damaged in a crash, there the SD Extreme has virtually eliminated the problem with UAVCAN enabled I have increased the log buffer to 5K now the log process actually always runs through max flight time is 8 minutes. With all other SD the error of the loss of the file system still exists with the other 2 PixRacers and in the PixFalcon. In beta and master 1.12 is another additional problem: as soon as UAVCAN is enabled, the buffer gets automatically reduced to 4k then the logger doesn´t start at all. It seems that the logger always needs at least 5K to start at all.
logger status
INFO [logger] Running in mode: all
INFO [logger] Number of subscriptions: 170 (5440 bytes)
INFO [logger] Full File Logging Running:
INFO [logger] Log file: /fs/microsd/log/2021-05-02/15_28_33.ulg
INFO [logger] Wrote 12.88 KiB (avg 0.21 KiB/s)
INFO [logger] Since last status: dropouts: 0 (max len: 0.000 s), max used buffer: 0 / 4396 B

davids5, 2021-05-05:
@dagar FYI:

>In beta and master 1.12 is another additional problem: as soon as UAVCAN is enabled, the buffer gets automatically reduced to 4k then the logger doesn´t start at all. It seems that the logger always needs at least 5K to start at all.

taileron, 2021-05-05:
With SanDisk Extreme 32G I can confirm there is no more file system failure at any moment.
This means, however, that these SD cards are not only recommended but the only ones that can be operated reliably in the PixRacer.  (fft and dshot is removed from the firmware cause currently I don´t need but dmesg added).

8K buffer size runs perfect in PixRacer if a SD Extreme is used, here a 25 minutes high rate log (with UAVCAN disabled):
free
             total       used       free    largest
Umem:       231312     183184      48128      47872
Prog:      2097152    1835008     262144          0
nsh> logger on
nsh> logger status
INFO  [logger] Running in mode: all
INFO  [logger] Number of subscriptions: 149 (4768 bytes)
INFO  [logger] Full File Logging Running:
INFO  [logger] Log file: /fs/microsd/log/2021-05-05/11_36_33.ulg
INFO  [logger] Wrote 0.91 MiB (avg 140.11 KiB/s)
INFO  [logger] Since last status: dropouts: 2 (max len: 0.004 s), max used buffer: 6575 / 8192 B
nsh> logger status
INFO  [logger] Running in mode: all
INFO  [logger] Number of subscriptions: 149 (4768 bytes)
INFO  [logger] Full File Logging Running:
INFO  [logger] Log file: /fs/microsd/log/2021-05-05/11_36_33.ulg
INFO  [logger] Wrote 2.42 MiB (avg 133.79 KiB/s)
INFO  [logger] Since last status: dropouts: 0 (max len: 0.000 s), max used buffer: 6548 / 8192 B
nsh> logger status
INFO  [logger] Running in mode: all
INFO  [logger] Number of subscriptions: 149 (4768 bytes)
INFO  [logger] Full File Logging Running:
INFO  [logger] Log file: /fs/microsd/log/2021-05-05/11_36_33.ulg
INFO  [logger] Wrote 167.74 MiB (avg 131.15 KiB/s)
INFO  [logger] Since last status: dropouts: 3 (max len: 0.014 s), max used buffer: 8183 / 8192 B
nsh> logger status
INFO  [logger] Running in mode: all
INFO  [logger] Number of subscriptions: 149 (4768 bytes)
INFO  [logger] Full File Logging Running:
INFO  [logger] Log file: /fs/microsd/log/2021-05-05/11_36_33.ulg
INFO  [logger] Wrote 192.43 MiB (avg 131.29 KiB/s)
INFO  [logger] Since last status: dropouts: 0 (max len: 0.000 s), max used buffer: 7739 / 8192 B
nsh> 
Buffer size changed to 5K (plattforms/nuttx/init) with UAVCAN enabled, there are drop outs but no lost file system:
free
             total       used       free    largest
Umem:       231312     212304      19008      18800
Prog:      2097152    1835008     262144          0
nsh> logger on
nsh> logger status
INFO  [logger] Running in mode: all
INFO  [logger] Number of subscriptions: 149 (4768 bytes)
INFO  [logger] Full File Logging Running:
INFO  [logger] Log file: /fs/microsd/log/2021-05-05/12_11_00.ulg
INFO  [logger] Wrote 1.29 MiB (avg 137.10 KiB/s)
INFO  [logger] Since last status: dropouts: 43 (max len: 0.010 s), max used buffer: 5101 / 5120 B
nsh> logger status
INFO  [logger] Running in mode: all
INFO  [logger] Number of subscriptions: 149 (4768 bytes)
INFO  [logger] Full File Logging Running:
INFO  [logger] Log file: /fs/microsd/log/2021-05-05/12_11_00.ulg
INFO  [logger] Wrote 38.72 MiB (avg 129.92 KiB/s)
INFO  [logger] Since last status: dropouts: 1217 (max len: 0.063 s), max used buffer: 4953 / 5120 B
nsh> logger status
INFO  [logger] Running in mode: all
INFO  [logger] Number of subscriptions: 149 (4768 bytes)
INFO  [logger] Full File Logging Running:
INFO  [logger] Log file: /fs/microsd/log/2021-05-05/12_11_00.ulg
INFO  [logger] Wrote 66.97 MiB (avg 130.10 KiB/s)
INFO  [logger] Since last status: dropouts: 797 (max len: 0.018 s), max used buffer: 5116 / 5120 B
nsh> logger off
nsh> sd_bench
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:  1368.20 KB/s, max write time: 18 ms (= 222.22 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run  1:  1389.66 KB/s, max write time: 23 ms (= 173.91 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run  2:  1380.26 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run  3:  1369.64 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run  4:  1440.07 KB/s, max write time: 5 ms (= 800.00 KB/s), fsync: 7 ms
INFO  [sd_bench]   Avg   :  1389.56 KB/s

mrpollo, 2021-05-05:
I have a PixRacer Pro and two different types of SanDisk microSD cards and some free time later today, I'm more than happy to help test what do you need @davids5?

[image]

taileron, 2021-05-05:
@mrpollo with the PixRacer Pro, the problem will probably not occur, as it is STM H743 with 64k buffer scheme without reduction in case of UAVCAN enabled. (Only F427 or its SD interface has the problem).
Exactly this 16G class 10 card picture left I now got new out of the box:
With the PixRacer R12 and R15, the file system disappeared at the first sd_bench attempt:
free
             total       used       free    largest
Umem:       231312     211696      19616      19248
Prog:      2097152    1835008     262144          0
nsh> sd_bench -r 35
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:  1951.53 KB/s, max write time: 38 ms (= 105.26 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run  1:  1686.92 KB/s, max write time: 60 ms (=  66.67 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run  2:  2121.45 KB/s, max write time: 35 ms (= 114.29 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  3:  2147.60 KB/s, max write time: 8 ms (= 500.00 KB/s), fsync: 5 ms
ERROR [sd_bench] Write error
nsh> cd /fs/microsd
nsh> ls
/fs/microsd:
nsh>

mrpollo, 2021-05-05:
Ok let me know if you need my help I have a plethora of hardware available

taileron, 2021-05-05:
For me personally, the problem with losing the file system is completely solved, because with SanDisk Extreme there is no problem at all and everything works with it in all my F427 FC. From now on I only need to keep the SanDisk Ultra and other SD cards away from the PixRacers.
In the 1.12 beta and master code the UAVCAN switched on 4K log buffer size does not work and has to be increased to at least 5K otherwise the logger doesn´start.
if param greater -s UAVCAN_ENABLE 1
then
	# Reduce logger buffer to free up some RAM for UAVCAN servers.
	set LOGGER_BUF 4

davids5, 2021-05-05:
@taileron @EliaTarasov 

Please test with https://github.com/PX4/PX4-Autopilot/pull/17533

You still may need to juggle the logger size for your configuration. But see if the logger and sd_bench feel more solid. Even with "Bad" cards or formats sizes that were not working before.

taileron, 2021-05-05:
@davids5 The fix solves it - but the sdio timeout needs not to be 6 times higher than before a factor of 1,5 (000fffff -> 1562500) is enough for stable writing on the following sd cards in the image.
[image]

EliaTarasov, 2021-05-06:
@davids5 @taileron 
Tested with #17533.

Still no luck to get sufficient enough log duration with SanDisk Extreme 32 GB. Only up to 5-7 minutes.
Have changed LOGGER_BUF to 5/10/15.
Have not seen considerable changes.

davids5, 2021-05-06:
On 

```
HW arch: PX4_FMU_V4
FW git-hash: ac6e7a1c6c62d12be6851e9ee7dfe823397d3914
FW version: 1.12.0 0 (17563648)
FW git-branch: master
OS: NuttX
OS version: Release 10.0.0 (167772415)
OS git-hash: 011fbb2ce150e05532e04b2f63d93ee932b226fb
Build datetime: May  5 2021 11:21:34
Build uri: localhost
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 0001000000003535323035375110003e0039
MCU: STM32F42x, rev. 3
```
with debugging enabled in nuttx (with the timing fix)

 I ran the  SanDisk Extreme 32 GB, 
36 minutes 40 seconds and had no issues related to the card.

After 1 hr 22 min - I ran out of free RAM. (@dagar, @bkueng I am assuming we have a memory leak)_

@EliaTarasov 
```
nsh>  logger status;date
INFO  [logger] Running in mode: all
INFO  [logger] Number of subscriptions: 141 (4512 bytes)
INFO  [logger] Full File Logging Running:
INFO  [logger] Log file: /fs/microsd/log/sess004/log001.ulg
INFO  [logger] Wrote 28.62 MiB (avg 27.29 KiB/s)
INFO  [logger] Since last status: dropouts: 0 (max len: 0.000 s), max used buffer: 5159 / 12288 B
Sat, Jan 01 18:43:08 2000
nsh>  logger status;date
INFO  [logger] Running in mode: all
INFO  [logger] Number of subscriptions: 141 (4512 bytes)
INFO  [logger] Full File Logging Running:
INFO  [logger] Log file: /fs/microsd/log/sess004/log001.ulg
INFO  [logger] Wrote 36.30 MiB (avg 27.28 KiB/s)
INFO  [logger] Since last status: dropouts: 0 (max len: 0.000 s), max used buffer: 5295 / 12288 B
Sat, Jan 01 18:47:56 2000
nsh>  logger status;date;sd_bench
INFO  [logger] Running in mode: all
INFO  [logger] Number of subscriptions: 141 (4512 bytes)
INFO  [logger] Full File Logging Running:
INFO  [logger] Log file: /fs/microsd/log/sess004/log001.ulg
INFO  [logger] Wrote 42.47 MiB (avg 27.28 KiB/s)
INFO  [logger] Since last status: dropouts: 0 (max len: 0.000 s), max used buffer: 5353 / 12288 B
Sat, Jan 01 18:51:48 2000
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench]
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:  1705.07 KB/s, max write time: 16 ms (= 250.00 KB/s), fsync: 1 ms
INFO  [sd_bench]   Run  1:  1691.09 KB/s, max write time: 18 ms (= 222.22 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  2:  1699.36 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run  3:  1699.36 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  4:  1671.71 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 8 ms
INFO  [sd_bench]   Avg   :  1693.30 KB/s
nsh> logger status;date;sd_bench
INFO  [logger] Running in mode: all
INFO  [logger] Number of subscriptions: 141 (4512 bytes)
INFO  [logger] Full File Logging Running:
INFO  [logger] Log file: /fs/microsd/log/sess004/log001.ulg
INFO  [logger] Wrote 48.36 MiB (avg 27.27 KiB/s)
INFO  [logger] Since last status: dropouts: 0 (max len: 0.000 s), max used buffer: 5240 / 12288 B
Sat, Jan 01 18:55:30 2000
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench]
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:  1709.74 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 5 ms
[... 43 lines cut ...]
INFO  [logger] Log file: /fs/microsd/log/sess004/log001.ulg
INFO  [logger] Wrote 94.85 MiB (avg 27.27 KiB/s)
INFO  [logger] Since last status: dropouts: 0 (max len: 0.000 s), max used buffer: 5489 / 12288 B
Sat, Jan 01 19:24:36 2000
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench]
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:  1673.40 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run  1:  1673.86 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  2:  1669.37 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run  3:  1681.85 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run  4:  1682.75 KB/s, max write time: 19 ms (= 210.53 KB/s), fsync: 5 ms
INFO  [sd_bench]   Avg   :  1676.24 KB/s
nsh>  logger status;date;sd_bench
INFO  [logger] Running in mode: all
INFO  [logger] Number of subscriptions: 141 (4512 bytes)
INFO  [logger] Full File Logging Running:
INFO  [logger] Log file: /fs/microsd/log/sess004/log001.ulg
INFO  [logger] Wrote 167.43 MiB (avg 27.26 KiB/s)
INFO  [logger] Since last status: dropouts: 0 (max len: 0.000 s), max used buffer: 5433 / 12288 B
Sat, Jan 01 20:10:04 2000
ERROR [sd_bench] Failed to allocate memory block
```

@EliaTarasov - could you have a bad card?

taileron, 2021-05-06:
I have also noticed that from a certain log buffer size of around 9K, sometimes the amount of free RAM is slowly and continuously reduced.

EliaTarasov, 2021-05-06:
@davids5 I could have, but i doubt. Checked it with `badblocks` and found nothing. The only thing that comes to mind is power supply. 
You noted previously:
> I find the SD card specs confusing, but I believe the "SanDisk Extreme 32Gb" is a UHS-{I|II} and is capable to run at 1.8V and high speed. All FMU HW does not support this! 3.3V only.

Maybe my particular FMU has some issues about that.

davids5, 2021-05-06:
@EliaTarasov , @taileron - I did a proper fix on F4,F7,H7 on the same #17533 (in https://github.com/PX4/PX4-Autopilot/pull/17533/commits/a7268742eee41c4fdb24f03bd7fc64da0d75937a) 

This ensures the SD drivers will always set the timeout to 250 Ms regardless of the system clocking. Before it was wrong and wronger :)

This value is a watchdog - were are not trying to optimize it. We are just trying to give cards the total time allotted in the SD spec

EliaTarasov, 2021-05-07:
@davids5 I tested #17533 , but result is approximately the same (depending on LOGGER_BUF size).
Not sure about the reason, have ordered another SanDisk Extreme and another Pixhawk 2.4.8.

taileron, 2021-05-07:
@davids5 With Omnibus F4, there was no problem despite an extremely slow SD, because it does not use SDIO but a SPI card interface. The SDIO on the F777 has not yet caused such a problem, regardless of the SD card, probably because X2.1-777 uses a separate sdio.c in Boards/SRC. As soon as I have time, I will also test it on the X2.1-777 to see if there are any negative repercussions, since it now runs perfectly without the sdio changes. 
For the solution of the F4 NUTTX sdio filesystem losses, #17533 should be merged as soon as possible cause it really solves it completely.

davids5, 2021-05-07:
@taileron @EliaTarasov - Thank you for testing!!! I have up-streamed the change and once approved I will back port them.

davids5, 2021-05-08:
@taileron @EliaTarasov - Again thank you all your testing!

This will be resolved when https://github.com/PX4/PX4-Autopilot/pull/17533 comes in shortly

The reason the bug appeared, is that 11 months ago, in the upgrade to NuttX 10.0.0, we enabled MUTLI BOLCK IO to the SD cards. This increased the cards busy time, depending on write load, format, age, etc, The delay in the F4 driver was just too short to accommodate the worse case multi-block write. F7/H7 Alwaus worked because minimum was correct (but the used values we too big as mentioned above.)