PX4/PX4-Autopilot #17455: Unable to flying more than 20 minutes

Kim-zhi-jiang, 2021-04-25:
Hello everybody

Unable to flying more than 20 minutes at v-master version in simulation environment.

1 MC iris
[image]

2 standard vtol
[image]

julianoes, 2021-05-07:
Interesting, this is probably something hanging due to lockstep.

Is this in VirtualBox? And how many CPU cores have you enabled? I'm just asking so I can try to reproduce this.

Kim-zhi-jiang, 2021-05-07:
Only one Ubuntu system in laptop, But V-1.11.x no this problem.
Test PX4 master version every day, still this problems (Don't know why the latency is high )

Kim-zhi-jiang, 2021-05-07:
@julianoes 

Thank you！  

Have been used before "PX4_SIM_SPEED_FACTOR=3 make -j4 px4_sitl gazebo"
If "PX4_SIM_SPEED_FACTOR=1" like this
[image]

Kim-zhi-jiang, 2021-05-07:
[image]

SalimTerryLi, 2021-05-08:
Confirmed with latest master f15eefcc9503d2abe0713defe7f017519e337b0b .

`make px4-sitl gazebo` and crashed after 30min. Ubuntu 20.04 host without any virtualization.

Seems that something went wrong within gzserver.

Jaeyoung-Lim, 2021-05-08:
@Kim-zhi-jiang Can you check your memory usage?

julianoes, 2021-05-08:
I can reproduce it. It doesn't seam to happen with jmavsim.

Jaeyoung-Lim, 2021-05-08:
@julianoes Thanks, I will have a look

Kim-zhi-jiang, 2021-05-08:
@Jaeyoung-Lim  Sometime RTL or TIMEOUT used today "541697d19" version.

[image]
[image]
[image]

julianoes, 2021-05-09:
Fix coming in https://github.com/PX4/PX4-SITL_gazebo/pull/747.

Kim-zhi-jiang, 2021-05-09:
@julianoes  @Jaeyoung-Lim 
Perfect！

closed 2021-05-09 by Kim-zhi-jiang

heleidsn, 2021-05-15:
Hi @julianoes and @Kim-zhi-jiang, 

Is this problem solved after the **PX4/PX4-SITL_gazebo#747**? 

I find the same problem and now I cannot simulate over 35min with `export PX4_SIM_SPEED_FACTOR=20`. It can be reproduced on my PC.

[image]

OS: Ubuntu 18.04
PX4: 5f14ee2a9705d746de7a87034f347db1c624fab2
px4_sitl: 2496b85dd22d2f9c1c1281d5ef77dfb86ef604a3

julianoes, 2021-05-17:
@heleidsn does this happen consistently?

Kim-zhi-jiang, 2021-05-17:
"PX4_SIM_SPEED_FACTOR=20"  so fast！I use 3
Some code needs to be optimized.

for(i=0 i++ i=1000000000){
double xyz=11111.0f;
.....
}
VS
double xyz;
for(i=0 i++ i=1000000000){
xyz=11111.0f;
.....
}

julianoes, 2021-05-17:
@Kim-zhi-jiang like where?

Kim-zhi-jiang, 2021-05-17:
I just touch drones,but find some, later will find EKF POS ATT.
[image]
[image]

heleidsn, 2021-05-24:
> @heleidsn does this happen consistently?

Hi @julianoes, this happens every time on my PC. 

Hi @Kim-zhi-jiang, as for `PX4_SIM_SPEED_FACTOR=20`, I think it's not too fast because it will reach the hardware limitation if we set `PX4_SIM_SPEED_FACTOR` a big number. On my PC, it can reach 20 times of real-time speed.

julianoes, 2021-05-25:
@heleidsn sorry, what happens?

heleidsn, 2021-06-01:
> @heleidsn sorry, what happens?

@julianoes Hi, I mean the same crash problem happens. As I said here: https://github.com/PX4/PX4-Autopilot/issues/17455#issuecomment-841588584

Kim-zhi-jiang, 2021-06-02:
@heleidsn 

More than this?

param set-default IMU_GYRO_RATEMAX 400 # Default 400 100 250 400 800 1000 2000Hz
param set-default IMU_INTEG_RATE 200 #Default 200 100 200 250 400Hz

Annoying lazy data transfer!

PR #17554 (2021-05-09): sitl_gazebo: update submodule
This fixes the simulation stalling after 30mins due to an int overflow.

Fixes #17455.