PX4/PX4-Autopilot #17471: Error: no autostart file found

tejassom, 2021-04-28:
Hi there,
I am trying to run PX4 simulator on MacOS - 11.3 (BigSur). 
Its failing at "make px4_sitl jmavsim"

with below error:
INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
Info: found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 10016
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG1_ID: curr: 0 -> new: 197644
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
Error: no autostart file found (Projects/Orion/developer/Firmware/build/px4_sitl_default/tmp/rootfs/etc/init.d-posix/airframes/10016_*)
ERROR [px4] Startup script returned with return value: 256
Buildfile: /Users/tejass/Winjit Work/Live Projects/Orion/developer/Firmware/Tools/jMAVSim/build.xml

Request you to help asap.

Thanks

vitalii-bulyzhyn, 2021-08-09:
Are there any updates on this issue? Have the same problem

vitalii-bulyzhyn, 2021-08-09:
SOLVED

There is an issue with folder naming: your root folder name should be named with one word. In my case it was "Drone Simulator" that didn't work, and issue gone after I renamed it to "drone_simulator"

P.S. to make it work you need to delete folder, create a new one and install simulator one more time. If you just rename folder - it won't work

BrendonLJJ, 2021-09-22:
Hi I have the same error but my folder name has no spacing, any idea whats wrong?

Buildfile: C:\users\brendon lee\px4_autopilot\Tools\jMAVSim\build.xml
Error: no autostart file found (lee/px4_autopilot/build/px4_sitl_default/tmp/rootfs/etc/init.d-posix/airframes/10016_*)
ERROR [px4] Startup script returned with return value: 256

vitalii-bulyzhyn, 2021-09-22:
@BrendonLJJ rename folder "brendon lee" to "brendon_lee", should work

BrendonLJJ, 2021-09-22:
Ah okay took me an hour to figure out how to change the user profile folder hahah
But I changed it and now it runs fine. Thank you so much!