PX4/PX4-Autopilot #17474: Warnings during VTOL transitions

AndreRigi, 2021-04-28:
**Current behavior**
In offboard, calling a VTOL transition ends up with PX4 [[FlightModeManager]](https://github.com/PX4/PX4-Autopilot/blob/d62204856cf0711af9ca82a399aac529783c412b/src/modules/flight_mode_manager/FlightModeManager.cpp#L179) running the fligh task `[FlightTaskTransition]`. This flight task [sets the position setpoints](https://github.com/PX4/PX4-Autopilot/blob/d62204856cf0711af9ca82a399aac529783c412b/src/modules/flight_mode_manager/tasks/Transition/FlightTaskTransition.cpp#L57) to `NAN`.
 
As the control mode remains in offboard (`_control_mode.flag_control_offboard_enabled` is true),
`[fw_pos_control_l1]` [expects a finite position setpoint](https://github.com/PX4/PX4-Autopilot/blob/d62204856cf0711af9ca82a399aac529783c412b/src/modules/fw_pos_control_l1/FixedwingPositionControl.cpp#L1704), which is not the case during transitions due to the previous flight task overwriting the setpoint.

Because the setpoint is not finite, the following warning is sent:
`[fw_pos_control_l1] Invalid offboard setpoint. `

However everything works fine as roll and pitch setpoints are override (standard vtol) during transitions.

**Expected behavior**
No warning during transitions.

**Possible solution?**
A quick and dirty solution could be to mute the warning during a transition, either by looking at the current flight task or by checking the vehicle status. We may overlook important warnings in some corner cases though...
A cleaner alternative may require an important change in the logic during transitions. 

**Log review**
[Log review](https://logs.px4.io/plot_app?log=1f88d1db-87e1-41db-9e8b-e90b4237c385) of an offboard VTOL flight with multiple transition.

ThomasRigi, 2021-05-03:
[image]
There are some issues with the velocity setpoint generation. I guess that this is caused by the MC transition flight task being overwritten by FW logic. So maybe fixing the underlying logic would also make the warnings go away

ThomasRigi, 2021-05-04:
I flew a real drone yesterday on v1.12.3-beta3(-ish) as always with sending SET_POSITION_TARGET_GLOBAL_INT messages, controlling only lat/lon/alt. Same issue with the changing Z setpoints, only that now I clearly noted that it really phyiscally affects as it also leads to oscillating MC thrust:
[image]
It was clearly audible that something wasn't quite right. 

For comparison, the same drone running the same offboard script on v1.11.2:
[image]

ThomasRigi, 2021-05-04:
@dagar @Jaeyoung-Lim this seems to be a follow-up issue from https://github.com/PX4/PX4-Autopilot/pull/16739 which survived https://github.com/PX4/PX4-Autopilot/pull/17130

Initially we thought it was only a false warning (and didn't check in depth :/), but now with visible observations on real flight it would be good to have this fixed